wled/WLED #4539: Failed to connect with static ip on WT32-ETH01

DarthWeber, 2025-02-04:
### What happened?

Hi, it looks like there might be an issue with the static ip configuration on a WT32-ETH01 board. I disabled WiFi to be sure it works with Ethernet. There is no problem with my network 192.168.178.0/24

After changing to 192.168.180.0/20 it is not reachable with that static ip  and it looks like it still get the ip from my DHCP:
[image]

to verfiy it did the same on a Wroom32 with wifi only. There everything works as expected:
[image]

Could there be a problem with static ip on a WT32-ETH01 esp. with CIDR 20?

Thanks,
Tom

### To Reproduce Bug

see above. Only with WT32-ETH01

### Expected Behavior

Static ip on ethernet with CIDR 20

### Install Method

Self-Compiled

### What version of WLED?

0.16.0-alpha (daily build)

### Which microcontroller/board are you seeing the problem on?

Other

### Relevant log/trace output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

DarthWeber, 2025-02-10:
Hm. But i only have this issue with Ethernet - not with WiFi

netmindz, 2025-06-14:
You are still using a subnet bigger than a mask of 255.255.255.0 however