wled/WLED #4544: Cannot build pov_display (user mod) for Esp32dev...

happysmartlight, 2025-02-09:
### What happened?

[image]

Cannot build image for esp32dev, esp32s3...

However, build good for esp32c3

[image]

### To Reproduce Bug

Help me fix it

### Expected Behavior

Help me fix it

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.15

### Which microcontroller/board are you seeing the problem on?

ESP32

### Relevant log/trace output

```shell
Processing esp32S3_wroom2 (platform: https://github.com/tasmota/platform-espressif32/releases/download/2023.06.02/platform-espressif32.zip; board: esp32s3camlcd; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
*** use existing my_config.h ***
npm run build
'npm' is not recognized as an internal or external command,
operable program or batch file.
*** Error 1
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32s3camlcd.html
PLATFORM: Espressif 32 (2023.6.2) > ESP32S3 CAM LCD
HARDWARE: ESP32S3 240MHz, 320KB RAM, 16MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-builtin, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 2.0.9
 - tool-esptoolpy @ 1.40602.0 (4.6.2) 
 - tool-mklittlefs @ 1.203.210628 (2.3)
 - tool-mkspiffs @ 2.230.0 (2.30)
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ strict
Found 42 compatible libraries
Scanning dependencies...
Dependency Graph
|-- AsyncTCP @ 1.3.1
|-- AnimatedGIF @ 1.4.7
|-- GifDecoder @ 1.1.0+sha.bc3af18
|-- FastLED @ 3.6.0
|-- IRremoteESP8266 @ 2.8.2
|-- NeoPixelBus @ 2.8.0
|-- ESPAsyncWebServerWLED @ 2.4.0+sha.2d3af2c
|-- QuickEspNow @ 0.6.2+sha.aa61f26
|-- arduinoFFT @ 2.0.1
|-- PNGdec @ 1.0.3
|-- Wire @ 2.0.0
|-- SD @ 2.0.0
|-- SD_MMC @ 2.0.0
[... 117 lines cut ...]
wled00/../usermods/pov_display/usermod_pov_display.h: In function 'void draw(PNGDRAW*)':
wled00/../usermods/pov_display/usermod_pov_display.h:42:5: error: 'png' was not declared in this scope
     png.getLineAsRGB565(pDraw, usPixels, PNG_RGB565_LITTLE_ENDIAN, 0xffffffff);
     ^~~
wled00/../usermods/pov_display/usermod_pov_display.h: In function 'uint16_t mode_pov_image()':
wled00/../usermods/pov_display/usermod_pov_display.h:55:14: error: 'png' was not declared in this scope
     int rc = png.open(filepath, openFile, closeFile, readFile, seekFile, draw);
              ^~~
wled00/../usermods/pov_display/usermod_pov_display.h: At global scope:
wled00/../usermods/pov_display/usermod_pov_display.h:67:23: error: in-class initialization of static data member 'const char PovDisplayUsermod::_data_FX_MODE_POV_IMAGE []' of incomplete type
     static const char _data_FX_MODE_POV_IMAGE[] PROGMEM = "POV Image@!;;;1";
                       ^~~~~~~~~~~~~~~~~~~~~~~
Compiling .pio\build\esp32S3_wroom2\lib7d3\IRremoteESP8266\IRtimer.cpp.o
*** [.pio\build\esp32S3_wroom2\src\usermods_list.cpp.o] Error 1
==================================================== [FAILED] Took 16.52 seconds ====================================================

Environment     Status    Duration
--------------  --------  ------------
esp32S3_wroom2  FAILED    00:00:16.522
=============================================== 1 failed, 0 succeeded in 00:00:16.522 ===============================================
```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

DedeHai, 2025-02-09:
does https://github.com/Aircoookie/WLED/pull/4427 fix this?

happysmartlight, 2025-02-09:
@DedeHai no... usermod only build ok for esp32c3 (however i cannot see it work)

DedeHai, 2025-02-09:
@Liliputech can you confirm this issue?

Liliputech, 2025-02-19:
Hi, the PR 4427 should be solving this issue, I'll check this out and provide new commit if it doesn't

hungvitourist, 2025-10-26:
hiện tại nó vẫn không được giải quyết