wled/WLED #4553: "Force max brightness" option in sync settings persists when live data has been overridden

Lyall-A, 2025-02-14:
### What happened?

I have Force max brightness enabled so WLED cannot control the brightness when using E1.32, but if I choose to override the live data I still cannot control the brightness even though I'm now controlling over WLED directly. I'm not sure if this is intended behaviour

### To Reproduce Bug

Sync Settings > Force max brightness: on
Connect to WLED using E1.32
Override in WLED
Try control brightness

### Expected Behavior

The brightness should change

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.15.0 (build 2412100)

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Relevant log/trace output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

blazoncek, 2025-02-15:
I think I can confirm (by looking at the code).

Please update led.cpp to have this in place of `applyBri()` and report back.
```c++
//applies global temporary brightness (briT) to strip
void applyBri() {
  if (realtimeOverride || !(realtimeMode && arlsForceMaxBri))
  {
    strip.setBrightness(scaledBri(briT));
  }
}
```

Lyall-A, 2025-02-15:
Thanks, I'm trying it right now but I'm having problems uploading the compiled firmware:
```
Uploading .pio\build\esp8266_2m\firmware.bin
Traceback (most recent call last):
  File "C:\Users\Lyall\.platformio\packages\tool-esptoolpy\esptool.py", line 34, in <module>
    import esptool
  File "C:\Users\Lyall\.platformio\packages\tool-esptoolpy\esptool\__init__.py", line 41, in <module>
    from esptool.bin_image import intel_hex_to_bin
  File "C:\Users\Lyall\.platformio\packages\tool-esptoolpy\esptool\bin_image.py", line 16, in <module>
    from intelhex import HexRecordError, IntelHex
ModuleNotFoundError: No module named 'intelhex'
```
I've never compiled WLED myself, is there something I'm missing? I did `pip install intelhex`

Lyall-A, 2025-02-15:
Flashed it manually, yes this does fix the issue