wled/WLED #4554: Controller crashes when using incorrect Content-Length in JSON request

someoverflow, 2025-02-14:
### What happened?

I’m using a Siemens SPS to control the LEDs on an ESP32-POE-ISO. Since it’s a Siemens SPS, I wrote a small program for the WLED JSON API, but I encountered an issue where the ESP controller crashes whenever I send an incorrect Content-Length header. Initially, I thought the issue was something entirely different because my program was timing out. However, after noticing that the controller’s uptime resets every time a request is made, it became clear that the timeout was actually due to the crash. I then started experimenting with the headers and found that if I omit the Content-Length header entirely, nothing happens so as long as the Content-Length is correct, it responds with a success JSON. But if I specify an incorrect value, the controller crashes.

Tried it on ESP32-POE-ISO and ESP32 C3

### To Reproduce Bug

Send a request like this to any controller, and the controller will crash, causing the web request to fail or timeout:

```HTTP
POST /json HTTP/1.1
Host: 192.168.1.235 (Controller IP)
User-Agent: Test Client
Content-Type: application/json
Content-Length: 10

{"on":true,"bri":128} 
```

### Expected Behavior

I’m not sure if this is fixable or if it’s an issue with the standard system packages, but I would expect a 500 Internal Server Error or some other error, rather than the controller crashing. I know it’s not a typical issue, but without access to logs or something that indicates the issue, troubleshooting becomes really challenging.

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.15.0 (2412100)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Relevant log/trace output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

blazoncek, 2025-02-15:
@willmmiles that may be something to look for in AWS?

willmmiles, 2025-02-17:
AsyncJSON, it looks like.  I was going to make a comment about un-vendoring all these deps under active maintainance elsewhere ... but actually it looks like this bug is still upstream, too.

DedeHai, 2025-08-31:
any progress on this with recent updates?