wled/WLED #4557: Playlists always append an "on:true"

DarthWeber, 2025-02-15:
### What happened?

My goal is to realize a secure relay timer on an GPIO (external multirelay pin). I did a playlist like:

> {"playlist":{"ps":[201,202],"dur":[10,10],"transition":[0,0],"repeat":1,"end":0,"r":false}}

but after saving this to the presets it will be appendend by an "on:true", but i don't want the main power to come up too.

### To Reproduce Bug

see above

### Expected Behavior

a playlist without "on:true"

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.15.0

### Which microcontroller/board are you seeing the problem on?

ESP32

### Relevant log/trace output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

DedeHai, 2025-02-15:
> it will be appendend by an "on:true", but i don't want the main power to come up too.

this is the default behaviour. why cant you just delete the "on:true" from the playlist if you dont need it?

DarthWeber, 2025-02-15:
That is the isdue: If i delete that part it will be added again after vsave

DedeHai, 2025-02-15:
ah I see, can confirm that. I do not know the reason it is that way but it looks like it is intentional.

blazoncek, 2025-02-15:
It is intentional.

dosipod, 2025-02-16:
Some of the guys reported that multiple times on discord and not really sure of the intended reasoning for it as if you 
want the unit to be on then that could be done in the presets but not the main playlist .This prevented a lot of the 
possibility for semi automation .

netmindz, 2025-02-16:
> It is intentional.

Why are we forcing on:true if the user did not request?

DarthWeber, 2025-02-16:
Thanks, that is what i also thought. Playlists could be more flexible wirhout forcing on the main output.

Maybe move to feature requests?

blazoncek, 2025-02-16:
> if the user did not request

User did request. In a lost post on Discord, quite a while ago.
I do not care if someone now requests removal of this feature. I will surely not do it but someone else might.
Expect going back and forth with it.

DarthWeber, 2025-02-16:
Would be great to understand the reason instead of moving back and forth.

In my eyes rhe json of a playlist should be saved as you modified it - with or wirhout that "on" statement

netmindz, 2025-03-16:
> > if the user did not request
> 
> User did request. In a lost post on Discord, quite a while ago. I do not care if someone now requests removal of this feature. I will surely not do it but someone else might. Expect going back and forth with it.

sorry, lazy wording. When I said user i was meaning that if the user had not selected something in the UI to set on:true, rather than a feature request from a user

blazoncek, 2025-03-16:
Let me put it other way:
- user creates presets without saving brightness
- user creates playlist of such presets
- user selects such playlist and WLED is in off state
- nothig happens

WLED's playlist work only if WLED state is on at the start of playlist.

DarthWeber, 2025-03-16:
Right. But that would work since the presets contains only settings like multirelays--Gesendet von meinem C64

blazoncek, 2025-03-16:
> But that would work since the presets contains only settings like multirelays

No, it wouldn't.

netmindz, 2025-03-22:
> Let me put it other way:
> 
> * user creates presets without saving brightness
> * user creates playlist of such presets
> * user selects such playlist and WLED is in off state
> * nothig happens
> 
> WLED's playlist work only if WLED state is on at the start of playlist.

Yeah the use/abuse of presets and playlists for things other than LED output does rather muddy the waters a bit. 

I think you are probably right that in that scenario users might expect the lights to turn on even if the playlist entries themselves were saved without brightness.

Adding a tick box to the UI for the playlist would be the ideal solution, but feels low priority. If others wish to submit a PR, great, but agree this isn't a bug

closed 2025-03-22 by netmindz