wled/WLED #4564: Fading transition on external segment renaming

kaese2002, 2025-02-17:
**Is your feature request related to a problem? Please describe.**
No Problem so far

**Describe the solution you'd like**
Im controlling a display with iobroker with changing segments names. I did not find a solution to fadeout and fadein from one message to the next.

**Describe alternatives you've considered**
No alternative solution...right now its a hard transition...simply changing text

**Additional context**

Thank you for your ideas for making WLED better!

Thanks a lot.
Regards

blazoncek, 2025-02-17:
It can't be done. At least not the way current implementation goes.
Someone would need to rewrite the effect from scratch (and modify WLED core).

DedeHai, 2025-04-28:
Soon...

blazoncek, 2025-04-28:
It does work.

kaese2002, 2025-04-28:
How? Small tutorial?