wled/WLED #4567: Matrix icons Database

Rushmed, 2025-02-18:
Hi Genteleman!

First I like to thank you for this project and all the hard work and time you spend.

Recnetly I bought an ESP32 8x8 matrix that I want to implement in my smarthome system to show message icons.
Unfortunately setting up icons as presets is not as easy as I expected.
Are there any plans to implement an icon library like [https://developer.lametric.com/icons](https://developer.lametric.com/icons) as AWTRIX and [PixelIt](https://pixelit-project.github.io/PixelIt/webui/#/gallery) does?
As I just want to use the 8x8 matrix for showing icons  AWTRIX an PixelIt are ineligible cause they reqire a 32x8 matrix.
So maybe there is a way of implementing Lametric DB in Pixel Magic Tool and making setting up icons in presets easy?

Thanks in further!

Jusius, 2025-02-19:
i think you can make a presets using tools that is provided by the community 

https://github.com/ajotanc/PixelMagicTool
https://github.com/werkstrom/WLED-PixelArtConverter

blazoncek, 2025-02-19:
Most likely no as resources on ESP are limited.
You are free however to explore other ways among which is a recent addition of GIF player and local storage.

Rushmed, 2025-02-19:
Okay, thanks!

closed 2025-02-25 by Rushmed