wled/WLED #4569: Support Smart LED Curtain

psxde, 2025-02-19:
I have tried to control an [Smart Curtain Style light from China](https://de.aliexpress.com/item/1005007965961621.html) via WLED but it seems to use a custom protocol for the different columns.

There are two variants which are shipped as the same product:

1. The LEDs all seem to be connected in parallel and have hardcoded addresses. There is no problem using WLED.
2. Each column seem to have 20x "dumb" WS28x LEDs and there is a logic element in each T-Connector at the top

I will continue talking about the second variant.

[image]

Each column is connected like that: Data In is connected to the 20 LED strip down and also to a LED inside the T part. The Output of this LED then is connected to the next T part. The LED inside this T-Connector seems not to light up but do some magic with the data.

With WLED only the first column lights up as expected but the other lights are shifted in some way.
- The second column begins with half of the brightness with green and blue, then follows the red channel of the previous led (in this case blank), then the second LED with green and blue, followed by the first LED with red, followed by the third LED with green and blue and so on
- The third column reacts with full brightness but splitted red of the first LED, continued by green and blue of the second LED
- All columns excluding the first one is blinking every few seconds and sometimes also changes the color

So the magic LED in the T-Connector seems in some way destroy the WS28x protocol.

Does someone have an idea how to support this type of led strip?

Thanks,
Patrick

blazoncek, 2025-02-20:
If there will be a NeoPixelBus LED type then it might be implemented in WLED.
Until then you will need to discuss with NeoPixelBus author for implementation.

psxde, 2025-02-21:
Found some more information: 
- https://wled.discourse.group/t/has-anyone-reverse-engineered-how-led-curtain-hooks-work/7423
- https://github.com/Makuna/NeoPixelBus/issues/790 

And the idea of using an Attiny 13 to build a replacement for the weird led circuit: https://forum.arduino.cc/t/ws2812-ic-to-drop-led-nodes/1342405

So without doing hardware changes to the LED curtain it seems to be impossible for now to get it to work with wled

psxde, 2025-03-08:
> I have similar set, but that is not the behaviour I see.
> 
> If I just set the led count to double what it should be then all LEDs light up in their expected colour.
> 
> I haven't got round to it yet, but in theory using an ledmap should correct for the missing pixels

I have get a third type of curtain which reacts like you described. 0-19 ist the first column, 39 the LED in the hook, 40-59 the next column. This curtain has transparent hooks which are wider than high as that ones which has hardcoded addresses.

But the other type with white hooks which are higher than wide, the protocol seems to be different.

AmexHusky, 2025-11-22:
I actually tested the TM1814 protocol as well, just to rule it out — but it doesn’t work on these curtains either. Even with WLED set to TM1814, the board doesn’t output anything usable.

The only setup that works reliably is feeding the data into the input of the so-called “level shifter” and using APA106 as the LED type in WLED. With that combination the curtain runs fine, except for the very last LED which keeps blinking — but that’s a minor issue and everything else behaves correctly.

So at this point it’s safe to say the curtain is not WS281x and not TM1814.
The custom data format clearly comes from that mystery “level-shifter” chip, and APA106 happens to be the only compatible mode WLED provides.

somebuddy87, 2025-11-22:
Thank you so much for this tip! I had only been doing my reverse engineering behind the original controller. So I was still running the original controller’s output signal through the logic analyzer — and PulseView even interpreted that as a WS280X signal. Because of that, I didn’t question it any further.

Could you please confirm once more which pin you are feeding the WLED signal into? I quickly measured the one shown below. It works, but especially during transitions I’m getting a lot of flickering on the curtain. I’ll definitely shorten the wires again to avoid any potential sources of error.

Attached is an image — could you please confirm that I picked the correct pin?

[image]

somebuddy87, 2025-11-22:
So… I did some more experimenting. With the APA106 setting, I get that flickering. With the 400 kHz profile, the flickering is gone, but the “first row” doesn’t respond. I worked around it by using the 400 kHz profile and connecting the first row to a different data pin on the ESP, driving it directly with the WS280x profile. This way, I now have all 400 LEDs working. I’m still struggling a bit with the matrix configuration in WLED, but I think that’s not a hardware issue.

braicc, 2025-11-27:
What wled release are you using? I find out that 15.1 it's working the way you describe.With 15.0 it's ok. I find the out after using a sp530x with wled firmware.

ZillionPrey, 2025-11-28:
> i have the exact same type from a store called action in germany.
> 
> The exact same behaviour as OP has described happens to me aswell. I have noticed that if you setup as 400 LED 20x20 Matrix and set the protocol to 400khz it seems to work to a certain extend.
> 
> The first row only glows white, and the others seem to react as they should. At least if yoh play matrix effects you can see them work properly.

@orichienal you can get almost all of them except the first row working by doing that :)

orichienal, 2025-11-28:
Hi guys, 
thx @braicc , @ZillionPrey , @psxde for the quick responses.

I tried every possible setting, basically everything once.

Now i installed version 15.0 and configured 400 kHz with 400 LEDs, 20 white and then 380 adjustable, but it behaves very strangely.

Sometimes I can set a color and it stays, and sometimes it flickers all the time. Or I can't set a certain color at all. And after I tried to adjust the brightness, it just went completely haywire.

So I thought I'd pull the plug and do a reset.
And now nothing works except for the first 20 that glow white.

Even playing around with the software via OTA didn't make any difference.

I then installed 15.0 via cable and now 380 are flashing wildly. Sometimes even one of the LEDs in the boxes at the top.

I have a good oscilloscope here and tried to analyze the signal from the original controllers, but unfortunately it's been too long since I've used such a device. If anyone has any tips, I'd be happy to hear them.

I'm not ready to reinstall the original controller yet. At most, I'll try a C3 mini, because the latest nightly build tells me that the memory isn't enough for 400 LEDs.

Have a wonderful day, evening, or night, everyone.

braicc, 2025-11-29:
Sorry to hear that. Once again I use 0.15.0 from web installer. I think after that something has changed in APA106 library. @psxde 

[image]
[image]
I get sometime  flickering  but I think that from mi cable 3x0.75 5meters. I use it whit Xlights.

bat86, 2026-10-03:
> Sorry to hear that. Once again I use 0.15.0 from web installer. I think after that something has changed in APA106 library( 0.15.0 is the first release with APA106 support). [@psxde](https://github.com/psxde)
> 
> I get sometime flickering but I think that from mi cable 3x0.75 5meters. I use it whit Xlights.
> 
> https://jumpshare.com/s/bkwulKPPqGnlcSCsuGXk

@braicc Huge thanks! It works absolutely flawlessly now!

The final piece of the puzzle to completely eliminate all flickering (especially on the red color and mixed shades) was adding a single WS2812B LED right after the ESP32 output pin, just before the curtain's connector.

In WLED settings, I set LED type to APA106, the correct matrix layout, and used "Skip first LEDs: 1". This single sacrificial pixel acted as a perfect 5V level shifter and cleaned up the signal timings. After that, all 400 pixels on the 20x20 curtain stabilized completely.

GreenAlienGoat, 2026-11-04:
Hi everyone, I stumble across your thread and hope you can help me with my setting (I'm pretty much a newbie with electronics).

I have this surplife curtain https://www.amazon.de/-/en/dp/B0D65GM692?ref=fed_asin_title, and a Gledopto 015wl with esp32. The first strand is working fine, but the rest is flickering like mad.

What I tried so far:

1. Sacrificial led. Didn't work
2. Plug in on powerbank and wall outlet. Nothing changed.
3. Attached the controller on a single 300 bead string. Works smooth.
4. Can't find the APA106 on my WLED app. It's greyed out on WS28xxx

Any idea would be apreciate!

DedeHai, 2026-11-04:
APA106 is supported. Also the new wled pixelbus driver allows for any timing. Closing.

closed 2026-11-04 by DedeHai