wled/WLED #4571: Multi Relay in AP Mode (/relays) not working?

pixelwave, 2025-02-20:
### What happened?

WLED v0.16.0-alpha "Kosen"
___

I use a WLED controller in AP only mode for a local installation without Wifi availability. 

I sent HTTP commands to that WLED controller (ESP32 WROOM) from a Remote (another ESP32) that is connected to that AP.

Commands like `http://4.3.2.1/win&A=255` work. 

Trying to toggle a relay with `http://4.3.2.1/relays?toggle=0,1,0,0` returns a 404 page.

When I connect the WLED controller to my local Wifi the command `http://192.168.178.59/relays?toggle=0,1,0,0` works as expected.

Is the endpoint `/relays` only available when connected to a WLAN?

Those two work even in AP mode:
`/json`
`/win`

### To Reproduce Bug

Put WLED controller in AP mode and send HTTPrequest to `/relays`

### Expected Behavior

HTTPrequest sent to `/relays` works in AP & Wifi Mode.

### Install Method

Self-Compiled

### What version of WLED?

WLED v0.16.0-alpha "Kosen"

### Which microcontroller/board are you seeing the problem on?

ESP32

### Relevant log/trace output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

blazoncek, 2025-02-20:
This behaviour is by design/original implementation.

[image]

Use JSON API if you want to control relays while in AP mode.

pixelwave, 2025-02-20:
Ah ok, thanks for clarification @blazoncek ! 

Would there be an "easy fix" by uncommenting or adding a line if I compile my own firmware?

blazoncek, 2025-02-20:
IDK, never needed that. If you find a solution, please submit a PR.

pixelwave, 2025-02-20:
Will try to figure something out. But I am far from an expert coder. x)

It feels a bit more like a bug than a feature that while `/json` and `/win` work in AP mode, `/relays` does not.

Has this something to do with usermods having more strict security guidelines?

closed 2025-06-28 by github-actions[bot]