wled/WLED #4572: не работает функция

ivan52605, 2025-02-21:
### What happened?

при подаче питания на мою esp8266 лента включается хотя чекбокс на включение при сбросе питания не стоит 

### To Reproduce Bug

обновил прошивку на последнюю

### Expected Behavior

лента не будет включатся при сбросе питания 

### Install Method

Self-Compiled

### What version of WLED?

WLED 0.15.0 

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Relevant log/trace output

```shell

```

### Anything else?

[image]

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

blazoncek, 2025-02-22:
English please.

ivan52605, 2025-02-22:
What happened?
when power is applied to my esp8266, the tape turns on, although the power on checkbox is not checked.

To reproduce the Error
, I updated the firmware to the latest

Expected Behavior
The tape will not turn on when the power is reset

Installation method
Self-compiling

What is the WLED version?
WLED 0.15.0

Which microcontroller/board do you see the problem on?
ESP8266

blazoncek, 2025-02-22:
Preset defines power state. Remove "on" key from preset.

ivan52605, 2025-02-23:
That is, turn off this checkbox.?

ivan52605, 2025-02-25:
Please help me, I still don't understand why this is happening, everything was fine before

ivan52605, 2025-02-27:
> Предустановка определяет состояние питания. Удалить клавишу "вкл" из предустановки.

Where can I find this preset?

wled-compile, 2025-03-13:
You have this option"Apply preset 25 at boot". Put "0" there instead of "25".