wled/WLED #4574: liveview not working

Ben00793, 2025-02-25:
### What happened?

I'm trying to access liveview, all I gwt is a black screen.

### To Reproduce Bug

I'm trying to access liveview, all I gwt is a black screen.

### Expected Behavior

To see the effects.

### Install Method

Binary from WLED.me

### What version of WLED?

V 0.15

### Which microcontroller/board are you seeing the problem on?

ESP32-S3

### Relevant log/trace output

```shell

```

### Anything else?

[image]

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

blazoncek, 2025-02-25:
You'll need a custom build (with JSON live view enabled) or switch to websockets.
This has been documented in changlog and release description.

closed 2025-02-25 by blazoncek

Ben00793, 2025-02-25:
How can I enable websockets?

blazoncek, 2025-02-25:
Either disable websockets (not recommended) or `#define WLED_ENABLE_JSONLIVE` in overrides.