wled/WLED #4583: PDM Microphone samples are not recognized at the input, Amplitude sits at 0

markusdd, 2025-03-02:
### What happened?

I have (new) boards sitting here with a TDK T3902 PDM MEMS Microphone and I am trying to get this to work.
ESP is an ESP32S3-N16R8.
I have tried WLED 0.15 as well my own build from latest/greatest master branch.

Issue I am observing: In PDM mode I am using GPIO4 as Clock out and GPIO5 as data in.
With an oscilloscope I can see the clock leaving the ESP and the data returning (yellow clock, blue data):

[image]

This gives me pretty high confidence that the hardware is fine.
Also, to exclude channel left/right woes, I reworked one board to have a left channel output and the other right channel output. The datasheet of the mic is here: https://www.lcsc.com/datasheet/lcsc_datasheet_2411220511_TDK-InvenSense-MMICT390200012_C3171752.pdf 
To me, this looks like a very 'run of the mill' PDM mic with hardly any specialties, the protocol is super simple. Rising edge right channel, falling edge left channel, that's it.

My pin config is as follows:

[image]

I also tried using WS as clk, same picture. I get proper output/input on the scope, but my console does this: (I have compiled with `  -D MIC_LOGGER  -D SR_DEBUG`):

[image]

Absolutely dead. I also tried forcing the data pin high from 3.3V, as this should result in a pegged high maximum volume. Also nothing.

When I configure it as a normal I2S Microphone, I do see this:

[image]

Level and Real sit around 30k constantly and volume is fluctuating. But there is zero 'real' reaction to music or clapping, which is logical, because this isn't a PCM signal but a PDM signal on the input, so essentially these numbers are random garbage, they will never be interpreted correctly in non-PDM mode.

I've tried everything I could think of, but I do not understand why there is zero reading of PDM data although it is there.

I tried skimming the C-Code of the audioreactive mod and there are numerous comments in the code regading PDM that I cannot make sense of, like this:

[image]

I do not assume we have an actual clock issue because it comes out.
I rather suspect these 2 (I have not touched any of the code, everything like it is on master except the debug defines):

[image]

So for some reason the data is not being sampled. I am at a loss for why.

Any guidance? 

### To Reproduce Bug

See above, but I guess you'd need such a mic or at least a PDM mic combined with an ESP32S3

### Expected Behavior

PDM data is being sampled and audioreactive works.

### Install Method

Self-Compiled

### What version of WLED?

0.15 and 0.16alpha@master

### Which microcontroller/board are you seeing the problem on?

ESP32-S3

### Relevant log/trace output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

markusdd, 2025-03-04:
When checking the log, the setup indeed looks as expected. I am at a loss as to why in proper PDM mode the data pin is seemingly not read.

Can anyone give guidance how to further debug?

```
Usermods setup
AR: I2S PDM Microphone - left channel only.
I2SSource:: initialize().
PIN ALLOC: Pin 4 successfully allocated by 0x20.
PIN ALLOC: 0x0000000000000lX.
PIN ALLOC: Pin 5 successfully allocated by 0x20.
PIN ALLOC: 0x0000000000000lX.
AR: I2S#0 driver without aPLL; fixed_mclk=0.
AR: 32 bits, Sample scaling factor = 0.2500
AR: I2S#0 driver installed in PDM MASTER mode.
FFT started on core: Total # of palettes: 0
0
Adding audio palettes.
1
2
3
```

blazoncek, 2025-03-05:
Perhaps a bug in Audioreactive usermod.
I know @softhack007 and @troyhacks were working on PDM microphone for MM fork.

troyhacks, 2025-03-05:
I've had success with PDM mics in WLED-MM with no major issues. 

Sometimes I've had to set ```-D I2S_USE_16BIT_SAMPLES``` in my builds or otherwise the audio is too low to be recognized. Quindor was doing some PDM testing the other day and he didn't have to add that tho.

@markusdd could you cross-test with WLED-MM mdev and see if that helps? 

WLED-MM has forked pretty far from mainstream WLED, but the AudioReactive fixes should be easy to backport if needed.

markusdd, 2025-03-05:
Ok, so after some more trying I am at this stage now:
[image]

This uses these settings as well as the `-D I2S_USE_16BIT_SAMPLES` define.

It only work with this (I have to take the legacy mode, normal PDM mode is dead):

[image]

So with these settings there seems to be something there, but very faint:

[image]

Also it seems to really well reactive to any music, so probably more noise floor than anything else, but at least not totally dead.

I had linked the datasheet of the Microphone I use above, any idea what might be different?
Maybe right instead of left channel? Any other weird settings?

markusdd, 2025-03-11:
Any thoughts from the experts on what to try next?

I think we have found modes where there certainly is an input being recognized but in any case whatever is there is way too low, even when holding the input high. So I assume it must be some magic setting in the software.

netmindz, 2025-03-11:
See recent discussion on PDM - https://discord.com/channels/700041398778331156/1076072513102168084/1339765830371639408

markusdd, 2025-03-12:
It says I do not have access to this.

closed 2025-07-18 by github-actions[bot]