wled/WLED #4586: ESP32-S3 and C3 hostname

b3-4r, 2025-03-04:
### What happened?

Hostname of WLED installed on C3 and S3 controllers is 'esp32s3-####', not 'WLED-servername' as is the case with older MCUs.

### To Reproduce Bug

Install WLED on an s3 or c3 board using the unofficial webinstaller.

### Expected Behavior

Hostname follows server description from UI settings page.

### Install Method

Self-Compiled

### What version of WLED?

WLED 0.16.0-alpha (build 2412040)

### Which microcontroller/board are you seeing the problem on?

ESP32-S3

### Relevant log/trace output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

blazoncek, 2025-03-06:
You are wrong. mDNS name is defined in WiFi settings, not UI settings.

b3-4r, 2025-03-07:
mDNS name is different from the hostname as reported to e.g. the DHCP server

blazoncek, 2025-03-07:
Only mDNS is used for communicating with device. Modified "serverDescription" as set for "hostname" is meaningless.

Rodney2K, 2025-03-07:
mDNS is different!

Start-Situation:

WiFi Settings:
mDNS -> http://      testmdns        .local
Client IP: 192.168.188.191
User Interface Settings:
server description:         WLED

After configuration save:
nslookup 192.168.188.191 ->
Server:  fritz.box
Address:  192.168.188.1
Name:    **wled-WLED**.fritz.box
Address:  192.168.188.191

 -> but after POWER CUT
nslookup 192.168.188.191
Server:  fritz.box
Address:  192.168.188.1
Name:    **esp32c3-f0cac5**.fritz.box
Address:  192.168.188.191

blazoncek, 2025-03-07:
Use mDNS explorer. It might be your Fritzbox is faulty and reports nDNS entries via DNS lookup. `nslookup` is used for regular DNS queries, not mDNS.

b3-4r, 2025-03-07:
.local is the mDNS address, the x.fritz.box addresses are FQDNs on a fritz.box network, construed as hostname.domain.tld, in their case by the fritzbox. Hostname or FQDN aren't usually something that can be used to access the webUI, rather a network-level identifier one would use e.g. to identify which devices are connected in a router's active DHCP client list.

mDNS address and network node hostname are distinct, this issue doesn't pertain to mDNS address

[image]

blazoncek, 2025-03-07:
Your best bet is to file a bug report with either Arduino framework or Espressif core WiFi/networking stack.
Or Fritzbox, BTW it has been known to cause issues.

On my own network, I do not see DNS addresses and DNS server is never populated with reverse or forward lookup from WLED devices. WLED does not explicitly do DDNS registration. All it does is advertises its presence via mDNS (2 TCP services, HTTP and WLED)

BTW FQDN are resolved by the DNS server from zone files. If you did not add WLED devices into zone files then DNS cannot resolve those addresses. mDNS on the other hand can as it uses multicast and not zone files. If Fritzbox resolves WLED devices via DNS then it is the issue of Fritzbox and its DNS implementation.

BBaoVanC, 2025-03-15:
I am having the same issue on cold boot! I am running Kea-DHCP4 for my DHCP, and the ESP is requesting `esp32c3-9ccba8` as the hostname on DHCP connect. Changing any of the Wi-Fi settings then "save & connect" fixes it, but if I do another cold reboot then it breaks again.

I have mDNS disabled since I do not want to use it whatsoever, but enabling it has no effect (as I would expect).

Using a QuinLED An-Penta-Mini which has an ESP32-C3. WLED 0.15.0 release, which was OTA updated from the stock firmware version (it is just unmodified WLED).

blazoncek, 2025-03-15:
Anyone interested in potential fix can check build artefacts from my fork. Click on commit hash above.

b3-4r, 2025-03-15:
> I am having the same issue on cold boot! 
> I have mDNS disabled since I do not want to use it whatsoever, but enabling it has no effect (as I would expect).
> I have a custom router setup so I can say for certain that it is the WLED device's fault for choosing the wrong hostname, not some different issue (I don't even deal with mDNS at all).

Same here on all 3 points.

BBaoVanC, 2025-03-31:
> Anyone interested in potential fix can check build artefacts from my fork. Click on commit hash above.

Just tried it and it doesn't seem to have worked on my ESP32-C3.

blazoncek, 2025-03-31:
Then this is beyond WLED's control.

schildbach, 2025-05-01:
I've seen this issue too with my ESP32-C3.

blazoncek, 2025-10-07:
@netmindz #4751 was closed without merging. I would call that "abandoned".