wled/WLED #4598: ArtNET proxyUniverse

cas22, 2025-03-16:
### What happened?

When trying to use ArtNet with the Max485 module you can't set the proxy universe under DMX Output to 0, as it disables the function. 
E1.31 starts on universe 1 while ArtNet starts on 0, so using anything but ArtNet will work.

### To Reproduce Bug

Compile WLED with `"WLED_ENABLE_DMX"`
It may be necessary to configure the sync interface to use ArtNet but the rest of the setting don't affect the outcome.
Under DMX Output >> Proxy Universe >> Set it to 0.
When you send and ArtNet signal nothing will happen.

### Expected Behavior

When ArtNet is selected as the Network DMX input, 0 should be an expected universe value

### Install Method

Self-Compiled

### What version of WLED?

WLED 0.16.0-alpha (build 2412040)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Relevant log/trace output

```shell

```

### Anything else?

The two places where this problem occurs are:

- In _dmx_output.cpp_ (void handleDMXOutput()) has `if (e131ProxyUniverse != 0) return;`
- And in _e131.cpp_ (void handleE131Packet()) has `if (e131ProxyUniverse > 0 && uni == e131ProxyUniverse)`

This to checks don't allow the universe to be 0, this could be fixed by adding a checkbox to disable the proxy mode instead of having it set to 0.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

netmindz, 2025-03-22:
0 is the magic value to disable, so by design rather than bug.

What happens if you use 1, do you get the first ArtNet universe or the second? The way the index is used within a universe is corrected for at a channel level so you might find the same is being done for the universe

netmindz, 2025-03-22:
p.s I would strongly advise against using this feature of WLED as it drops most of your updates anyway as it can't come close to the full speed of DMX

Ok for basic wash lights, useless for any moving heads, pixel mapping etc

cas22, 2025-03-24:
> 0 is the magic value to disable, so by design rather than bug.
> 
> What happens if you use 1, do you get the first ArtNet universe or the second? The way the index is used within a universe is corrected for at a channel level so you might find the same is being done for the universe

When you set it as 1 you get the second universe, instead of the first (0).
I understand it is by design but it is a mistake either way, as it should let you use it.

closed 2025-08-01 by github-actions[bot]