wled/WLED #4599: Usermod: cant create a dropdown menu.

artystudios-GH, 2025-03-17:
### What happened?

i have a hexagon table with 54 hexagons. each hexagon has a ledstrip with 16 leds and 1 touch sensor. i want a usermod that when i touch a hexagon, only that segment lights up. this part is working, but i also want to set the animation when i touch the sensor. so i need a dropdown menu with the names of the animations. see the code below

this does not work unfortunately, there is a drop-down menu but its empty. i tried every LLM available, but no luck.
can someone help me out? Or is this a bug in version 15?

### To Reproduce Bug

`#include "wled.h"
#include <FastLED.h>
#include "FX.h"
 
#define NUM_LEDS 864
#define DATA_PIN 4
 
int segments[54][2] = {{0, 15}, {16, 31}, {32, 47}, {48, 63}, {64, 79}, {80, 95}, {96, 111}, {112, 127}, {128, 143}, {144, 159}, {160, 175}, {176, 191}, {192, 207}, {208, 223}, {224, 239}, {240, 255}, {256, 271}, {272, 287}, {288, 303}, {304, 319}, {320, 335}, {336, 351}, {352, 367}, {368, 383}, {384, 399}, {400, 415}, {416, 431}, {432, 447}, {448, 463}, {464, 479}, {480, 495}, {496, 511}, {512, 527}, {528, 543}, {544, 559}, {560, 575}, {576, 591}, {592, 607}, {608, 623}, {624, 639}, {640, 655}, {656, 671}, {672, 687}, {688, 703}, {704, 719}, {720, 735}, {736, 751}, {752, 767}, {848, 863}, {832, 847}, {816, 831}, {800, 815}, {784, 799}, {768, 783}};
 
int Pins[4] = {17, 27, 26, 25}; // S0, S1, S2, S3
int MP4 = 33; // MUX kanaal 4
int MP1 = 34; // MUX kanaal 1
int MP2 = 35; // MUX kanaal 2
int MP3 = 32; // MUX kanaal 3
 
unsigned long epochTimes[64];
unsigned long delayTimes[64];
bool muxValues[64];
bool previousValues[64];
 
class usermod_Hexagon_Touch : public Usermod {
private:
  // Configuratie
  unsigned long delayDuration = 1000; // Tijd voordat segment uitgaat
  byte Speed = 200;                   // Snelheid van animatie (0-255)
  byte Animatie = FX_MODE_RAINBOW_CYCLE; // Standaard animatie
  bool enabled = false;               // Usermod aan/uit
  bool initDone = false;
 
  // Statische strings in PROGMEM
  static const char _name[];
  static const char _enabled[];
  static const char _delayDuration[];
  static const char _Speed[];
  static const char _Animatie[];
 
  // Verminderde lijst met animaties (20 opties)
  const char* wledAnimations[20] = {
    "Solid", "Blink", "Breathe", "Wipe", "Wipe Random", "Random Colors", "Sweep", "Dynamic", "Colorloop", "Rainbow",
    "Scan", "Fade", "Chase", "Chase Rainbow", "Running", "Twinkle", "Dissolve", "Sparkle", "Strobe", "Fireworks"
  };
 
public:
  void setup() {
    Serial.begin(115200);
    if (enabled) {
      pinMode(MP1, INPUT);
      pinMode(MP2, INPUT);
      pinMode(MP3, INPUT);
      pinMode(MP4, INPUT);
      pinMode(Pins[0], OUTPUT);
      pinMode(Pins[1], OUTPUT);
      pinMode(Pins[2], OUTPUT);
      pinMode(Pins[3], OUTPUT);
 
      for (int i = 0; i < 64; i++) {
        muxValues[i] = false;
        previousValues[i] = false;
        epochTimes[i] = 0;
        delayTimes[i] = 0;
      }
[... 170 lines cut ...]
### Install Method

Binary from WLED.me

### What version of WLED?

0.15.0

### Which microcontroller/board are you seeing the problem on?

ESP32

### Relevant log/trace output

```shell

```

### Anything else?

[image]

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

DedeHai, 2025-03-17:
first off, dont expect AI's to generate valid code.
second: the code is full of little things that are not done right (but may still work), like re-initializing serial, overriding pin definitions, just at a glance.
dropdowns do work. try back-stepping: make a very simple droptdown, maybe copy code from a working usermod. then increase complexity.

artystudios-GH, 2025-03-17:
a simplified code still doesn't work unfortunately.

`#include "wled.h"
 
 
 
class usermod_Hexagon_Touch : public Usermod {
private:
  // Configuratie
  unsigned long delayDuration = 1000; // Tijd voordat segment uitgaat
  byte Speed = 200;                   // Snelheid van animatie (0-255)
  byte Animatie = FX_MODE_RAINBOW_CYCLE; // Standaard animatie
  bool enabled = false;               // Usermod aan/uit
  bool initDone = false;
 
  // Statische strings in PROGMEM
  static const char _name[];
  static const char _enabled[];
  static const char _delayDuration[];
  static const char _Speed[];
  static const char _Animatie[];
 
  // Verminderde lijst met animaties (20 opties)
  const char* wledAnimations[20] = {
    "Solid", "Blink", "Breathe", "Wipe", "Wipe Random", "Random Colors", "Sweep", "Dynamic", "Colorloop", "Rainbow",
    "Scan", "Fade", "Chase", "Chase Rainbow", "Running", "Twinkle", "Dissolve", "Sparkle", "Strobe", "Fireworks"
  };
 
public:
  void setup() {
   
  }
 
  void loop() {
 
  }
 
 
  void addToConfig(JsonObject& root) {
    JsonObject HEXAGON_TOUCH = root[FPSTR(_name)];
    if (HEXAGON_TOUCH.isNull()) {
      HEXAGON_TOUCH = root.createNestedObject(FPSTR(_name));
    }
    HEXAGON_TOUCH[FPSTR(_enabled)] = enabled;
    HEXAGON_TOUCH[FPSTR(_delayDuration)] = delayDuration;
    HEXAGON_TOUCH[FPSTR(_Speed)] = Speed;
    HEXAGON_TOUCH[FPSTR(_Animatie)] = Animatie;
  }
 
  bool readFromConfig(JsonObject& root) {
    JsonObject top = root[FPSTR(_name)];
    if (top.isNull()) return false;
 
    enabled = top[FPSTR(_enabled)] | enabled;
    delayDuration = top[FPSTR(_delayDuration)] | delayDuration;
    delayDuration = constrain(delayDuration, 10, 10000);
    Speed = top[FPSTR(_Speed)] | Speed;
    Speed = constrain(Speed, 0, 255);
    Animatie = top[FPSTR(_Animatie)] | Animatie;
    Animatie = constrain(Animatie, 0, 19); // Aanpassen aan de nieuwe lengte (19 in plaats van 114)
 
    Serial.printf("Config loaded: enabled=%d, delayDuration=%lu, Speed=%d, Animatie=%d\n", enabled, delayDuration, Speed, Animatie);
 
    if (initDone) setup();
    return true;
  }
 
  void addToJsonInfo(JsonObject& root) {
    JsonObject user = root["u"];
    if (user.isNull()) user = root.createNestedObject("u");
 
    // Enabled toggle knop
    JsonArray infoArr = user.createNestedArray(FPSTR(_name));
    String uiDomString = F("<button class=\"btn btn-xs\" onclick=\"requestJson({");
    uiDomString += FPSTR(_name);
    uiDomString += F(":{");
    uiDomString += FPSTR(_enabled);
    uiDomString += enabled ? F(":false}});\">") : F(":true}});\">");
    uiDomString += F("<i class=\"icons ");
    uiDomString += enabled ? "on" : "off";
    uiDomString += F("\"></i></button>");
    infoArr.add(uiDomString);
 
    // DelayDuration input
    JsonArray delayArr = user.createNestedArray(FPSTR(_delayDuration));
    delayArr.add(F("<input type=\"number\" min=\"10\" max=\"10000\" value=\""));
    delayArr.add(String(delayDuration));
    delayArr.add(F("\" onchange=\"requestJson({"));
    delayArr.add(FPSTR(_name));
    delayArr.add(F(":{"));
    delayArr.add(FPSTR(_delayDuration));
    delayArr.add(F(":parseInt(this.value)}})\">"));
 
    // Speed input
    JsonArray speedArr = user.createNestedArray(FPSTR(_Speed));
    speedArr.add(F("<input type=\"number\" min=\"0\" max=\"255\" value=\""));
    speedArr.add(String(Speed));
    speedArr.add(F("\" onchange=\"requestJson({"));
    speedArr.add(FPSTR(_name));
    speedArr.add(F(":{"));
    speedArr.add(FPSTR(_Speed));
    speedArr.add(F(":parseInt(this.value)}})\">"));
 
    // Animatie placeholder
    JsonArray animArr = user.createNestedArray(FPSTR(_Animatie));
    animArr.add(F("<select onchange=\"requestJson({"));
    animArr.add(FPSTR(_name));
    animArr.add(F(":{"));
    animArr.add(FPSTR(_Animatie));
    animArr.add(F(":parseInt(this.value)}})\">"));
    animArr.add(F("</select>"));
  }
 
  void appendConfigData() {
    Serial.println(F("appendConfigData called"));
 
    // Voeg de dropdown toe
    oappend(F("addDropdown('Hexagon_Touch','Animatie');"));
    Serial.println(F("Dropdown added"));
 
    // Voeg de 20 animatie-opties toe
    char buffer[100];
    for (int i = 0; i < 20; i++) {
      snprintf_P(buffer, sizeof(buffer), PSTR("addOption('Hexagon_Touch','Animatie','%s',%d);"), wledAnimations[i], i);
      oappend(buffer);
      Serial.printf("Added option %d: %s\n", i, wledAnimations[i]);
      delay(1);
    }
 
    // Markeer de geselecteerde animatie
    if (Animatie >= 0 && Animatie < 20) {
      snprintf_P(buffer, sizeof(buffer), PSTR("selectOption('Hexagon_Touch','Animatie',%d);"), Animatie);
      oappend(buffer);
      Serial.printf("Selected option: %d\n", Animatie);
    }
 
    Serial.println(F("appendConfigData finished"));
  }
};
 
usermod_Hexagon_Touch Usermod_Hexagon_Touch;
 
const char usermod_Hexagon_Touch::_name[] PROGMEM = "Hexagon_Touch";
const char usermod_Hexagon_Touch::_enabled[] PROGMEM = "enabled";
const char usermod_Hexagon_Touch::_delayDuration[] PROGMEM = "delayDuration";
const char usermod_Hexagon_Touch::_Speed[] PROGMEM = "Speed";
const char usermod_Hexagon_Touch::_Animatie[] PROGMEM = "Animatie";`

closed 2025-03-17 by blazoncek

artystudios-GH, 2025-03-17:
i dont have or want a discord account.