wled/WLED #4603: Can't save more than 4 outputs

marcone, 2025-03-19:
### What happened?

I'm using a Wemos D1 Mini, ESP8266. According to the knowledge base WLED supports "Up to 3 LED outputs per ESP8266 instance", but I found I can actually define 4 (they're all PWM, so perhaps the limit of 3 applies to some other type?).
However, I cannot define more than 4. When I add a 5th output it lets me set its type, GPIO and other attributes, but it doesn't save it. There's no error message or any other indicator as to why it isn't saving it, it just doesn't.

At first I thought it might be because the number of PWM outputs is limited. The [technical reference](https://www.espressif.com/sites/default/files/documentation/esp8266-technical_reference_en.pdf)  isn't entirely clear on the matter, alternately talking about 4, 5, 8 and 16 PWM channels being supported. Even when I define the 5th one as a simple on/off non-PWM output though, wled still doesn't save it.
Basically, I can specify 4 outputs using any of the 5 available GPIOs (4, 5, 12, 13 and 14), but I can't use all 5.

### To Reproduce Bug

Add 4 "PWM White" outputs, choosing 4 GPIOs from (4, 5, 12, 13, 14).
Add a 5th output using the remaining GPIO.
Click the "Save" button.
Click "LED Preferences" again and note that the 5th output was not saved.

### Expected Behavior

Either it should save the 5th output, or tell me why it didn't.

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.15.0 (build 2412100)

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Relevant log/trace output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

marcone, 2025-03-20:
> Technically not a bug

It's still a UI bug. The UI lets me specify 6 LED outputs (the '+' disappears after adding 6 outputs), but only saves 4.

blazoncek, 2025-03-20:
It will allow you to specify up to 36 outputs.

marcone, 2025-03-20:
How? After I add the 6th output, the '+' button to add more outputs no longer appears.
I.e. after the 5th one I can still add one more by using the '+' button:

[image]

but after the 6th one there's no more '+' button:

[image]

blazoncek, 2025-03-20:
Sorry. In 0.16

blazoncek, 2025-03-22:
@marcone please test [this](https://github.com/blazoncek/WLED/actions/runs/14008917083/job/39226235154).
Select appropriate environment and use artefacts from there.

blazoncek, 2025-03-22:
@PaoloTK you are also invited to check if it breaks any of your environments.

marcone, 2025-03-22:
That lets me save 5 LED outputs now, but the 5th one doesn't actually work.

blazoncek, 2025-03-22:
@marcone workings of PWM output is beyond this change.
As for it @DedeHai or @willmmiles may know more why 5th PWM output wouldn't work on ESP8266. In any case IMO that would be of very low priority to fix.

marcone, 2025-03-22:
The 5th output also doesn't work if I make it on/off instead of PWM.

PaoloTK, 2025-03-24:
> The 5th output also doesn't work if I make it on/off instead of PWM.
You might want to compile with WLED_DEBUG flag and post the output when you hit save in the LED settings page.

I'm unable to test myself due to not having esp8266 devices (all our environments are esp32 based) but I can probably guide you through the process of finding the issue since I went through it for esp32 to fix 10+ outputs.

marcone, 2025-03-26:
For what it's worth, I compiled main branch WLED with WLED_MAX_BUSSES=5, and that both lets me save 5 outputs, and all 5 of them work.
When I have time I'll try @blazoncek's unified-blend branch and see why that 5th output wasn't working for me there.

closed 2025-07-12 by blazoncek