wled/WLED #4604: Wired DMX config and warning about lack of support both showing

netmindz, 2025-03-21:
### What happened?

Should be only one or the other, not both

### To Reproduce Bug

View sync settings with V4 and non-v4

### Expected Behavior

Only the right choice displayed - warning for non-v4, input for v4

### Install Method

Self-Compiled

### What version of WLED?

Dev

### Which microcontroller/board are you seeing the problem on?

ESP32

### Relevant log/trace output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

harueg, 2025-05-09:
I have a usermod for wired DMX in, when anybody cares.

netmindz, 2025-05-10:
@harueg we now have dmx in as a core feature, this is just about a UI bug where we are not enabled and disabling certain ui elements

harueg, 2025-05-10:
Hi
Here it is anyway, Maybe there is something useful in it. Usermod is Filmbauten.

[Filmbauten-StreetLamps-Receiver-main.zip](https://github.com/user-attachments/files/20139232/Filmbauten-StreetLamps-Receiver-main.zip)