wled/WLED #4607: color wheel shows for plain white LEDs

marcone, 2025-03-23:
### What happened?

Not sure if this is a documentation or a code bug, or me misunderstanding.
(I asked on Discord, but got no response)

I have defined 4 outputs, all set to "PWM White".
According to [the settings documentation](https://kno.wled.ge/features/settings/), the "Auto-calculate white channel from RGB" setting "_only appears if "Type" is set to a type that has more than three color channels_".
Despite me having specified only PWM White outputs, the auto-calculate white channel setting appears.
Since the setting is there, I set it to "Max", which according to [the doc on white handling](https://kno.wled.ge/features/cct/) is "good for white-only LEDs without any RGB".
Back in the Colors tab of the web interface, this causes the color wheel/picker to show, even though I only have white LEDs.
If I set the auto-calculate white channel setting to "None", then I only get a "white channel" slider in the Colors tab, which I have to make sure to set to the maximum value so I can then use the brightness slider to actually control brightness.

### To Reproduce Bug

- Define only "PWM White" outputs
- Note that the "Auto-calculate white channel from RGB" setting is available, which is contrary to what the [documentation](https://kno.wled.ge/features/settings/#:~:text=only%20appears%20if%20%22Type%22%20is%20set%20to%20a%20type%20that%20has%20more%20than%20three%20color%20channels) says.
- Set that setting to Max, which the [documentation on white handling](https://kno.wled.ge/features/cct/#:~:text=good%20for%20white%2Donly%20LEDs%20without%20any%20RGB) says is "good for white-only LEDs without any RGB"
- save and go to the Colors tab in the web UI, and note that despite having defined only white LED outputs, the full color wheel/picker is shown

### Expected Behavior

I expect the web UI to not show an RGB color picker for white LEDs.

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.15.1-b1 (build 2503200)

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Relevant log/trace output

```shell

```

### Anything else?

When I open the javascript console in Chrome, I can see that the `hasRGB` variable has a value of 1, while the `hasWhite` variable has a value of 0 when the "Auto-calculate white channel from RGB" setting is set to Max. When set to "None", `hasRGB` is 0 as expected, and hasWhite is 1 as expected.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

blazoncek, 2025-03-23:
If you select "Auto calculate White from RGB" other than "None" WLED will treat your white output ar RGB, hence Color wheel.

closed 2025-03-23 by blazoncek

marcone, 2025-03-23:
So it's a documentation bug then?

blazoncek, 2025-03-23:
Documentation may be a bit outdated (0.13). You are welcome to help with it.

marcone, 2025-03-24:
After thinking about this some more, I think the real issue is that "PWM White" is not considered a white-only LED light, which is why the "Auto-calculate white channel" option is shown.

This seems odd to me (I mean why treat a single output called "PWM White" as part of an RGB light?), but appears to have been done on purpose, as comments in the code say that `TYPE_ANALOG_1CH` (which maps to the string "PWM White"), is "one channel (white), relies on auto white calculation".

blazoncek, 2025-03-24:
You are overthinking this a bit. 😉 

One thing you need to know is that WLED started as a RGB(W) controller and only recently got full support for independent white-only output.

The current behaviour is in line with legacy behaviour while also featuring newer approach. Not every comment or documentation was updated to reflect that. Still, as with any software, bugs may be present but IMO this is not one of them.

Documentation is community based and relies on contributions from people like you or anyone else, so if you think you can improve it, please do.