wled/WLED #4610: "Auto-calculate W channel from RGB" appears to be ignored for last output

marcone, 2025-03-24:
### What happened?

The value of the "Auto-calculate W channel from RGB" options for each of the defined outputs determines whether the "Colors" tab shows a color wheel or not, but it appears this is ignored for the last output in the segment.

### To Reproduce Bug

- define 4 "PWM White" outputs
- set the "Auto-calculate W channel from RGB" option for the first output to "Max", and for the other 3 outputs to "None"
- save and go back to the Colors tab, and note that the color wheel is shown, as expected.

But if instead of the first one, you set the LAST one to Max, then the color wheel is NOT shown.

### Expected Behavior

Since all of the outputs are in the same segment, it shouldn't matter _which_ output has the "Auto-calculate W channel from RGB" set to "Max". If any one of them has it set that way, the color wheel should be shown.

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.15.0 (build 2412100)

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Relevant log/trace output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

blazoncek, 2025-03-25:
Please provide JSON status & info message from WLED for different configuration. Use your browser for that, developer tools, network tab.

marcone, 2025-03-25:
Not sure what exactly you mean by "JSON status & info messages", but I've attached the response for /json/si after setting it to Max/None/None/None (which shows the color wheel), and after setting it to None/None/None/Max (which does not show the color wheel).
All 4 outputs work either way, but the UI is different for those two cases.

[max-none-none-none.txt](https://github.com/user-attachments/files/19451365/max-none-none-none.txt)

[none-none-none-max.txt](https://github.com/user-attachments/files/19451366/none-none-none-max.txt)

PR #4658 (2025-06-01): Segment layers and better effect transitions (blending)
Segment layering & effect blending improvements
- Photoshop-style segment/layer blending
- return of strip ABL
- remove global LED buffer in favour of segment-local buffer
- new effect blending modes/transitions
- custom palettes moved out of WS2812FX class
- increased limits (matrix size, LED RAM)
- added "rainbow"-mode colorwheel
- replaced palettes with gamma unmodified ones
- move gamma adjustment to last step before sending to LEDs
- Segment & WS2812FX class reorganisation (mutable members, reordered members, protected members)

Fulf [... cut]
closed 2025-06-01 by blazoncek