wled/WLED #4628: Support NEW Esp32-S3 boards

RiWi-creator, 2025-04-02:
**Is your feature request related to a problem? Please describe.**
Development boards Unexpected Maker ProS3 & Adafruit QT PY S3 
I can install Wled on both Development boards, however when restart(reset), the boards keep rebooting over and over again and no access point will appear.

**Describe the solution you'd like**
Support for the new ESP32-S3 boards

**Describe alternatives you've considered**
Working WLED source code i can use with these boards

**Additional context**
Technical information of the two development board:
Unexpected Maker ProS3
ESP32-S3 Dual 32bit Xtensa LX7 cores @ up to 240Mhz, 512KB SRAM
External Flash & SPRAM
WINBOND W25Q128JVSQ 16MB Flash (Quad SPI)
ESP-PSRAM64H 8MB PSRAM (Quad SPI)

Adafruit QT PY S3
ESP32-S3FH4R2 Dual Core 240MHz Tensilica processor, 512KB SRAM
4MB Flash (Quad SPI), 2MB PSRAM (Quad SPI)

Thank you for your ideas for making WLED better!

DedeHai, 2025-04-04:
S3 is generally supported, did you try https://wled-install.github.io/ with correct settings?

MrBeepoint, 2025-04-07:
Maybe your issue is related to your pin configuration? Which pins do you use?

blazoncek, 2025-04-07:
Source code is adjusted and fully supports S3 chips (all of the variants).
You need correct binary however for particular combination of RAM, PSRAM and flash.
Please use WLED [forum ](https://wled.discourse.group)or [Discord](https://discord.gg/FMvz8nzmat) for help and support questions.

closed 2025-04-07 by blazoncek

RiWi-creator, 2025-04-07:
This is the schematic of the board i im using.
https://github.com/UnexpectedMaker/esp32s3/blob/main/schematics/schematic-pros3.pdf