wled/WLED #4629: WLED 0.15+ always resets GPIO 0 to become a button pin (two new bugs)

Arcitec, 2025-04-02:
### What happened?

The ability to have more than 2 buttons has introduced a bug in WLED's pin handling.

- The `WLED_MAX_BUTTONS` default array length is dynamic and is 2 for ESP8266 and 4 for ESP32. It can also be edited by the compiler flag to become even larger if wanted.
- WLED only gives 2 default values for `BTNPIN` and `BTNTYPE`. Which means there's some unspecified values in the array on ESP32:
```cpp
#define BTNPIN 0,-1
#define BTNTYPE BTN_TYPE_PUSH,BTN_TYPE_NONE
```
- C++ arrays default all unspecified values to 0. So `btnPin[4] = {2,3}` becomes `{2,3,0,0}` automatically. And for example, WLED's default values for an ESP32 binary will look like this:
```cpp
int8_t btnPin[4] = {0, -1, 0, 0};
byte buttonType[4] = {2, 0, 0, 0};
```
- **Bug 1:** The WLED code that is supposed to fix this at the first launch in `cfg.cpp: if (fromFS) {` never executes. That code is supposed to set `GPIO = -1 and Type = NONE` for every button pin whose pre-defined firmware setting for the GPIO was -1 or type was NONE. Basically a "reset all unused pins". But that code never executes!
- This means that all of the unused/unspecified button pins all stay defined as "GPIO 0 type NONE".
- **Bug 2:** There's a second bug in the button initialization code: It activates and takes over ownership of ALL "defined" (`>= 0`) button GPIO pins _even_ if their type is set to NONE (Disabled). So it steals/allocates all "disabled" button pins too. If a button setting is "disabled", it should not steal ownership of them!
- And because the button initialization code runs _after_ the LED initialization, it means that **GPIO 0** is *always* overwritten to become a button pin instead of an LED pin on all ESP32 devices.
- The most important bug is that the settings cleanup code never executes. But the "pins are taken over by the button handler even if their type is set to NONE (Disabled)" is a nasty bug too.

There is a temporary workaround via compiler flags: By manually specifying every array index on ESP32 builds:

```ini
-D WLED_MAX_BUTTONS=4
-D BTNPIN=0,-1,-1,-1
-D BTNTYPE=2,0,0,0
```

### To Reproduce Bug

x

### Expected Behavior

x

### Install Method

Binary from WLED.me

### What version of WLED?

0.15 and newer

### Which microcontroller/board are you seeing the problem on?

ESP32

### Relevant log/trace output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Arcitec, 2025-04-02:
I've been digging around the issue tracker now. There has been previous discussion of this and some talks about fixes, but it didn't have a good summary of the actual bugs this causes, so it got lost in the noise.

- Closed issue: Talks about unset button pins becoming 0 by default, but didn't realize that it was a big problem causing LED outputs to be overwritten. https://github.com/wled/WLED/issues/3953
- Closed pull request: Attempted to "fix" it by hardcoding a list of `-1` values, which was not a good solution since the length of the button array can be changed dynamically. https://github.com/wled/WLED/pull/4074
- Stale pull request: Outdated code and inappropriate solution which attempted to add a "init to -1" in the wrong place in the code: https://github.com/wled/WLED/pull/4098

The last pull request above actually talks about the `cfg.cpp` solution.

However, if we start digging into v0.15.1.beta2, we see some interesting changes in WLED:

https://github.com/wled/WLED/tree/v0.15.1.beta2

https://github.com/wled/WLED/blob/6572efbf9f8a977160b19f01fc48df24ccc8c434/wled00/cfg.cpp#L292-L299

There's code which is supposed to set the buttons to `GPIO = -1, Type = NONE (0)` if either GPIO was -1 or Type was NONE (0).

@PaoloTK told me on Discord that the new code doesn't run. He said that if we add a debug print statement to the highlighted code block, it actually never runs, so the fix doesn't work?

Here's what to do next:

1. Check if the highlighted fix above actually runs and actually works.
2. If it works, close this and all the other linked PRs and issues.

PaoloTK, 2025-04-05:
> @PaoloTK told me on Discord that the new code doesn't run. He said that if we add a debug print statement to the highlighted fromFS code block, it actually never runs, so maybe the fix doesn't work?

To clarify, this was my experience when testing while working on the previous 2 PRs. It could have been fixed since then, but I have been unable to test.

blazoncek, 2025-04-17:
@Arcitec how familiar are you with git patches? I can provide a patch for testing but is unfortunately based on my fork's `stable` branch.
You'll need to add remote to your local git repo, checkout my branch and apply patch on top of that. Then do the testing.

netmindz, 2025-08-16:
Has this been resolved now @blazoncek  ?

blazoncek, 2025-08-16:
> Has this been resolved now?

Once #4757 is merged.
However I have not received any confirmation or other feedback on the PR. Neither form @Arcitec nor @PaoloTK or anyone else.

netmindz, 2026-03-03:
> This is happening again in latest version

0.15.3 or the nightly builds of 16?