wled/WLED #4634: AJ-SR04M sensor doesn't work with Animated Starcase usermod

lexxnsk, 2025-04-08:
### What happened?

Hello.
I have AJ-SR04M with 2 pins. It works fine with ESP8266 and some basic Code from Arduino IDE, but doesn't work with WLED Animated Staircase usermod. At the same time regular HC-SR04 works fine with WLED. I really appreciate if you can help me. Thanks.

### To Reproduce Bug

Connect AJ-SR04M sensor  - doesn't work.
replace with HC-SR04 - works

### Expected Behavior

Works without any differences

### Install Method

Self-Compiled

### What version of WLED?

0.15

### Which microcontroller/board are you seeing the problem on?

ESP32

### Relevant log/trace output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

blazoncek, 2025-04-08:
Does it say this sensor is supported?
If not, you'll need to contact usermod creator and request support.

lexxnsk, 2025-04-08:
@willmmiles hello, may I ask you to have a look?

lexxnsk, 2025-04-08:
@realrolfje hello, may I ask you to have a look?

lexxnsk, 2025-04-09:
#4637 https://github.com/wled/WLED/pull/4637

lexxnsk, 2025-04-09:
Hello. I've fixed the issue. Please, have a look and approve this PR https://github.com/wled/WLED/pull/4637
Thanks.