wled/WLED #4636: W281x CCT support

justr0st, 2025-04-09:
**Is your feature request related to a problem? Please describe.**
I stumbled upon a nice WS2811 based CCT lightstrip [like this](https://amzn.eu/d/1HnvAHK). It is connected as follows:
- RED WS2811 output&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;-> WW LED
- GREEN WS2811 output&nbsp;&nbsp;&nbsp;&nbsp;-> CW LED

Where #ff0000 color means full warm white and #00ff00 is full cold white.

**Describe the solution you'd like**
It would be great to add a WS281x CCT option to support both W and C channels on RGB IC outputs. Adding "W281x CCT" output type would solve the problem.
I imagine the W and C channels could be both swappable with any of R, G and B outputs for maximum compatibility.

**Describe alternatives you've considered**
Alternative would be to stick with PWM CCT lightstrip which require MOSFET PWM outputs and not all controllers have those.

**Additional context**
There are multiple manufacturers that seem to produe the same or similar type of led strip:
https://amzn.eu/d/gggzlQM
https://amzn.eu/d/1bz2pNr
https://aliexpress.com/item/1005005515009828.html
https://aliexpress.com/item/1005007799233087.html

DedeHai, 2025-04-09:
Try WS281x WWA

justr0st, 2025-04-09:
Did not realize I wasn't on beta so the output type was missing. Will test that later today

justr0st, 2025-04-09:
@DedeHai just tested and it works perfectly, thanks for the tip. The issue can be closed.
The enhancement request is similar to #552 however my kind of led strips does not have any amber output, only WW and CW. So i don't know if my request is a duplicate or not.

closed 2025-04-09 by justr0st