wled/WLED #4638: Custom preset is not applied after toggling wled power state

kudp02, 2025-04-09:
### What happened?

I created a custom preset that does not rely on effects. This preset is applied on boot, but when I toggle the WLED power state manually afterward, the selected effect (could be any) takes precedence over this preset.

### To Reproduce Bug

1. Create preset via json:
2. `{"on":true,"bri":235,"seg":{"id":0,"i":[0,"fca203",1,4,"fca202",4,7,"fca203",7,9,"fda303",9,11,"fca203",11,13,"fda303",13,15,"fca203","fca202",16,19,"fca203",19,21,"fda303","fca203",22,27,"fda303",27,29,"fca203","fda303",30,33,"fca203",33,35,"fda303","fca203","fda303","fca203",38,41,"fda303","fca203","fca202","fca203",44,46,"fda303",46,54,"fca203",54,56,"fda303","fca202","fca203",58,60,"fda303","fca203","fca103",62,64,"fca203","fca103",65,68,"fca203","fda303",69,74,"fca203","fc9e02","fc9802","fc9503","fc9403","fc9803","fc9e03","fca103","fda303",82,87,"fca203","fb9b03","fb8d02","fa8302",90,92,"f97c02","fa8302","fb8c02","fb9a02","fca203","fda303",97,99,"fca203","fba102","fb9c02","fa8a02","f87801","f96a01","f86101","f96201","f96a01","f97702","fa8802","fb9a03","fca103",111,113,"fca203","fba002","fb9302","fa7d02","f86701","f85500","f74900","f74901","f85401","f96602","f97a02","fb9102","fba003","fca103","fda303","fb9f03","fb8d02","f97502","f75d01","f74700","f63600","f63601","f74601","f85b01","f97201","fb8c02","fb9e03","fca103","fca203","fca003","fb8d02","f97502","f75d01","f64701","f63601","f63501","f64601","f75b01","fa7201","fb8c02","fb9e03","fca103","fca203","fca103","fb9302","fa7c02","f86702","f75501",160,162,"f64901","f75401","f96501","fa7902","fa9002","fb9f03","fda203","fca203","fba103","fb9b03","fa8a02","fa7702","f96a02",174,176,"f96201","f86901","fa7702","fb8902","fb9a02","fba103","fda303",182,185,"fca203","fc9a03","fa8b02","fa8002",188,190,"fa7b02","fb8103","fb8b03","fc9803","fca102",194,196,"fca202",196,199,"fda303","fca103","fc9d03","fc9803",202,204,"fc9403","fc9703","fc9e03","fda203",207,212,"fda303",212,215,"fca203",215,217,"fda203",217,219,"fca202",219,225,"fda303",225,229,"fca203",229,231,"fda303",231,233,"fca202",233,235,"fca203",235,237,"fda303","fca203","fda303",239,259,"fca203",259,261,"fda303",261,271,"fca203",271,275,"fda303",275,280,"fca203"]}}`
3. Enable this preset
4. Turn off wled (do not unplug)
5. The selected effect should be applied instead of this preset

### Expected Behavior

After toggling the power state, the last preset should be applied and take precedence over the effect. 

### Install Method

Binary from WLED.me

### What version of WLED?

v0.15.0 "Kosen"

### Which microcontroller/board are you seeing the problem on?

ESP32-C3

### Relevant log/trace output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

blazoncek, 2025-04-09:
If you are using 0.15 or later, check UI settings.

kudp02, 2025-04-09:
Needed to set `Power button preset override for On:`

closed 2025-04-09 by kudp02