wled/WLED #4640: Fading White dose not work, Its only Updated once wen Fade is at 100% and yumps to its endvalue

patrick-dmxc, 2025-04-11:
### What happened?

Hi, i use SK6812 Stripes
And in Advanced Mode the Stripe is only fading the RDB-LEDs but not the White LEDs,

[image]

[image]

[image]

### To Reproduce Bug

I fade from Blue to white whats happening is Blue LED is fading to 0% the white LED stays at 0% untill Fade is over then it jumps to 100%

### Expected Behavior

I fade from Blue to white whats wold i see is Blue LED is fading to 0% the white LED Fades to 100% 

### Install Method

Binary from WLED.me

### What version of WLED?

v0.16.0-alpha "Niji" (Build: 2412040)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Relevant log/trace output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

blazoncek, 2025-06-27:
Why do you have "White balance correction" and "Calculater CCT from RGB" selected?

DedeHai, 2025-06-27:
also that cct blending over 100% is a bug that was fixed in https://github.com/wled/WLED/pull/4737
@patrick-dmxc can you please update so we are sure to not chase old bugs?

blazoncek, 2025-06-27:
As per @Aircoookie 
- "White balance correction" its intention is to modify RGB values with "warmness" tint (1900K - 10060K) from CCT slider
- "Calculate CCT from RGB" will determine color temperature from RGB value and set CCT accordingly (in the range 1900K - 10060K)
- "Auto calculate white" will determine white channel brightness from RGB values (or allow manual if set so)

1st will add CCT slider in UI even if the strip has no CCT LEDs.
2nd will remove CCT slider from UI if strip has CCT LEDs
3rd will remove white channel slider from UI if Accurate or Brighter are selected and strip has W channel.