wled/WLED #4641: WS2811 "RGBCCT" Support

ForsakenHarmony, 2025-04-12:
**Is your feature request related to a problem? Please describe.**
This LED strip seems to have 2 WS2811 chips per segment, one for normal RGB, the other for CW + WW (R and G, B is blank)

https://www.aliexpress.com/item/1005007799677123.html

**Describe the solution you'd like**
Support LED strips that have multiple chips for one position?

**Describe alternatives you've considered**

**Additional context**

DedeHai, 2025-04-13:
There are jus too many variants of WS2811 to add support for them all. 
You are invited to add it and do a PR. If you can't you need to wait for someone to do so. If these type of stripes become very popular, get people to upvote.

jobisingh, 2025-10-02:
I'd definitely like to see this. I have about 60 of these strips and cannot get warm white to work under any of the other given options. I have seen some people have mild success with these strips using the FW1906 controller setting. However there is a lack of color accuracy that makes having such a nice strip worthless.

DedeHai, 2025-10-02:
For anyone who want support for this, please give the color order (RGB WW CW Blank for example or (GRB CW Blank WW) to check which ones are the most common.

jobisingh, 2025-10-02:
> For anyone who want support for this, please give the color order (RGB WW CW Blank for example or (GRB CW Blank WW) to check which ones are the most common.

Ideally in the following order, RGB + CW + WW

Mopsmaciek123, 2025-10-02:
@jobisingh Which brand of strip do You have? The one from BTF Lighting I have posted?

jobisingh, 2025-10-02:
I don't know if this is relevant for anyone, but the LED strip I and some others are referencing are ADRESSABLE strips. No idea if that is relevant.

so010, 2025-10-19:
Looks like these are the only stripes that check both boxes:

- [ x] FCOB

- [ x] RGB+CCT

matteocali, 2025-12-30:
The solution processed by @JATimmers works great for me with no start-up flashes. I didn't manage to test the fork of @darekxan but would be awesome to have a pull request so in the next release this kind of LED strips would be properly supported.

hybra, 2026-03-02:
I applied @darekxan changes to WLED 0.15, complied and patched the GLEDOPTO GL-C-017WL-D firmware with it. It updated successfully and my WS2811 RGBCCT strips work flawlessly now.  Be aware that in my case after patching the firmware I had to set "Use global LED buffer" to OFF in the Config. I also had to set the color order to RGB and divide the number of leds by 5 (so if for a normal strip output you set 120 here you have to set 24 in the Length field). Everything works as expected, including effects.

efaden, 2026-04-05:
> Thank you so much! I also own the GLEDOPTO GL-C-017WL-D with WS2811 RGBCCT and WLED version 0.16.0-alpha, and it generally works. However, when I turn it on, it blinks briefly and then dims to a higher brightness. If I briefly turn it off and on again, it doesn't blink. But if I wait about 10 seconds, it does blink. Do you have any idea what could be causing this?

How do you have it configured....?  I'm running 0.17-dev and can't for the life of me get anything but red. Or do I need to compile my own?

Nichtsahnender, 2026-04-06:
Like this:
[image]

But I compiled it on my own, I don't know if 0.17 has the "WS2811 2IC 5CH".