wled/WLED #4642: Make the segment brightness to work with every effect

Andro-Marian, 2025-04-13:
**Is your feature request related to a problem? Please describe.**
I try to add some effects as overlay but the segment brightness has no effect.

**Describe the solution you'd like**
To change the brightness of segments.

**Describe alternatives you've considered**
-

**Additional context**
To have a better contrast between Text and other effects.

DedeHai, 2025-04-13:
coming soon.
sneek preview here: https://github.com/wled/WLED/pull/4612

blazoncek, 2025-04-13:
That's a feature I'm developing. It is available in my fork already as a finished code. The PR mentioned above is a draft to get segment blending and particle system cooperate. Which do not ATM.