wled/WLED #4643: MQTT not updated when interacting with the HA integration

chatziko, 2025-04-15:
### What happened?

When making changes to WLED via the web interface, HTTP API (legacy or json) or MQTT itself, the updated values are published in the corresponding MQTT topics. However, when making any change via the Home Assistant integration, the MQTT topics are not updated at all.

This is always reproducible under both 0.15.1.beta2 and 0.14.4.

### To Reproduce Bug

- Enable MQTT
- Make changes to the lights via HA
- Observe the topics not getting updated

### Expected Behavior

Updates should always be published to MQTT, independently from where they came from.

### Install Method

Binary from WLED.me

### What version of WLED?

0.15.1.beta2

### Which microcontroller/board are you seeing the problem on?

ESP32

### Relevant log/trace output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

DedeHai, 2025-04-15:
AFAIK WLED has not special treatment for HA. If HA uses the API incorrectly, not much we can do.

blazoncek, 2025-04-15:
There are two aspects when MQTT message is not published: one is *cooldown* the other is WS only notification.
Cooldown prevents spamming interfaces with frequent updates. Will not change.
WS only notifications are intended to notify WS clients only. HA is not WS client AFAIK.

blazoncek, 2025-04-15:
This is intentional. "v" is verbose response, meaning client is interested in getting the state response.

blazoncek, 2025-04-15:
Not my code and can't judge *why* it was implemented so. Feel free to provide a patch.

blazoncek, 2025-07-14:
@chatziko please test #4771 and report back (in the PR).

blazoncek, 2025-08-02:
@chatziko final call or I'm pulling the plug as nobody else requested this.

chatziko, 2025-08-04:
Hi @blazoncek , I missed the PR, thanks a lot for working on this! This is _not_ in the nightly build, right? I need to build it myself. Not a problem, I just need to figure it out.

PR #4771 (2025-09-17): Fix for #4643 - Missing MQTT updates
Fixes #4643

## Summary by CodeRabbit

* **New Features**
  * MQTT state publication now occurs when verbose responses are requested via the JSON HTTP endpoint or WebSocket, provided MQTT support is enabled.

* **Bug Fixes**
  * Ensured MQTT updates are sent in sync with verbose API responses for improved state consistency across integrations.
closed 2025-09-17 by blazoncek