wled/WLED #4646: syslog / net debug - configurable via gui

KrX3D, 2025-04-16:
Hello,

currently using syslog /net debug you need to add

```
	#define WLED_DEBUG
	#define WLED_DEBUG_HOST "192.168.1.3"
	#define WLED_DEBUG_PORT 514
```

and than you can enable/disable it in the gui at info.

--

it would be nice if the syslog feature can be:

- 1. just added via i.e. #define WLED_NEW_DEBUG or #define WLED_SYSLOG, and the options above can be added as optional 
- 2. in the WLED GUI under /settings/sync it would be nice to 
-- Enable Syslog , maybe also under the INFO menu where you can currently enable/disable the normal debug
-- Option to Enter IP
-- Option to enter Port
- 3. It would be nice to either be able to be able to enable the syslog option AND the normal Serial output, OR be able to select which one. Or disable both

Thank you very much.

KrX3D, 2025-04-19:
i also noticed that disabling Debug in the gui:

[image]

and rebooting the devices, enables it again

blazoncek, 2025-04-19:
> rebooting the devices, enables it again

That is intentional. It is enabled by default.

KrX3D, 2025-04-19:
> > rebooting the devices, enables it again
> 
> That is intentional. It is enabled by default.

Hmm ok, for me i like to have it compiled with the option and than have it enabled whenever i want it.

blazoncek, 2025-04-19:
If debugging is implemented with QuickDebug (enhanced with net debugging) then having an option to permanently disable debugging from UI would make sense, otherwise it doesn't as you normally don't use debug version.

netmindz, 2025-04-20:
As far as I'm aware, WLED's net debug does not use the syslog protocol

KrX3D, 2025-04-26:
will be solved with 

https://github.com/wled/WLED/pull/4664

closed 2025-04-26 by KrX3D