wled/WLED #4647: Syslog / Net debug - several logs created because of missing hostname

KrX3D, 2025-04-16:
### What happened?

Hello, i use the syslog / net debug feature.

I can set those options in my syslog server for "System identifier for logfile name"

[image]

For the first and last everything is ok, but when i use "Hostname (from syslog message)" which i use mostly i get several log files created because the hostname is missing after the time

[image]

i.e.: syslog-Segment.log

[image]

### To Reproduce Bug

#define WLED_DEBUG
#define WLED_DEBUG_HOST "192.168.1.3"
#define WLED_DEBUG_PORT 514

and have a syslog server which uses the hostname from the send log after the date

### Expected Behavior

instead of have just i.e. 

`Apr 16 20:46:20 Segment geometry: 0,30 -> 0,1`

to have

`Apr 16 20:46:20 HOSTNAME Segment geometry: 0,30 -> 0,1`

this should create than just one log file with the complete log

### Install Method

Self-Compiled

### What version of WLED?

0.16.0-alpha

### Which microcontroller/board are you seeing the problem on?

ESP32-S2

### Relevant log/trace output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

netmindz, 2025-04-20:
From you description it sounds like the debug messages do not contain a hostname, hence why if you select the option on your syslog server to extract the hostname from the message you get unexpected results.

To be honest I was unaware the debug feature was even trying to implement the syslog protocol, i thought it was raw logs, where did you see reference to WLED using syslog?

blazoncek, 2025-04-21:
@netmindz you are correct. Netdebug does not implement any protocol, it is intended to transmit raw debug messages.
However if it can be made syslog compatible it wouldn't hurt.
Though I do not see areal value beyond using that feature wit multiple units being debugged simultaneously.

But just a note of warning: netdebug will not provide you with crash dumps.

KrX3D, 2025-04-26:
will be solved with 

https://github.com/wled/WLED/pull/4664

closed 2025-04-26 by KrX3D