wled/WLED #4648: async-mqtt-client compiled and linked in when WLED_DISABLE_MQTT is set

rhkean, 2025-04-17:
### What happened?

I just noticed that the async-mqtt-client is compiled and linked into the bin even when the WLED_DISABLE_MQTT flag is set.

by deleting the entire folder from the src/dependencies folder, the bin file is 3.2k smaller

### To Reproduce Bug

add -D WLED_DISABLE_MQTT flag to platformio.ini file
build the project
notice that the async-mqtt-client is still listed as being compile and linked

delete the async-mqtt-client folder from the src/dependencies folder
see that the bin is 3200 bytes smaller

### Expected Behavior

async-mqtt-client is not linked and/or included in the .bin file

### Install Method

Self-Compiled

### What version of WLED?

WLED 0.15.1.beta2

### Which microcontroller/board are you seeing the problem on?

ESP32

### Relevant log/trace output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

willmmiles, 2025-05-06:
> [@willmmiles](https://github.com/willmmiles) could you take a look what's causing it? I peeked into ELF and can see some AsyncMqttClient references but can't make anything useful from it.

The linker always includes .cpp files that are part of the project itself -- only those in static libraries are filtered based on whether or not there are external references.  The assumption is that you could have important work being done in static constructors, for example.

IMO the best fix is to un-vendor the library, and call upon the latest version from upstream via `lib_deps`; I'm pretty sure there's been quite a few fixes since it was vendored in here ...

The second-best fix is to move it to the "lib" folder, import a proper library.json (ideally from the upstream project), and add it to `lib_deps`.

Giving the dependencies some love is next on my list after usermod wrangling.  I've still got that old ArduinoJSON v7 branch kicking around, too...

blazoncek, 2025-06-27:
FYI upstream library marvinroger/async-mqtt-client has not been updated in 3 years.

rhkean, 2025-08-28:
it makes a difference if you disable mqtt... about 2k

DedeHai, 2025-09-03:
@netmindz the commit is not to main, still needs porting.

netmindz, 2025-09-03:
Thanks for the heads up @DedeHai , where is this commit then?

DedeHai, 2025-09-03:
its a link to Blaz dev branch I think.

closed 2025-09-04 by netmindz