wled/WLED #4649: Usermod to force LED reset in off state for long signal line interference

Fmajor, 2025-04-18:
**Hardware Setup:**
- ESP8266 controller
- long data line (~5 meters)

**Problem Description:**
When strip is off (bri=0):
1. First several LEDs randomly light up (with random color) due to signal interference
2. Requires manual toggle (off->on->off) to fully reset

**Usermod Requirements:**
- Force all LEDs to black every 60 seconds when they are off (bri=0)

**Current Implementation:**
```c
// usermod_force_off.h
#ifndef USERMOD_FORCE_OFF_H
#define USERMOD_FORCE_OFF_H

#include "wled.h"

class UsermodForceOffIfOff : public Usermod {
private:
  unsigned long lastCheck = 0;
  unsigned long intervalMs = 60000; // default 1 min

  bool enabled = true;
  int intervalSeconds = 60;

public:
  void setup() override;
  void loop() override;

  void addToConfig(JsonObject& root) override;
  bool readFromConfig(JsonObject& root) override;
  void addToJsonInfo(JsonObject& root) override;
};

#endif // USERMOD_FORCE_OFF_H

// usermod_force_off.cpp
#include "usermod_force_off.h"

void UsermodForceOffIfOff::setup() {
  intervalMs = intervalSeconds * 1000;
}

void UsermodForceOffIfOff::loop() {
  if (!enabled) return;

  unsigned long now = millis();
  if (now - lastCheck >= intervalMs) {
    lastCheck = now;

    if (bri == 0) {
      // also try strip.setBrightness(0);
      strip.fill(0)
      strip.show();
      Serial.println("[ForceOffUsermod] Forced off (bri == 0)");
    }
  }
}

void UsermodForceOffIfOff::addToConfig(JsonObject& root) {
  JsonObject obj = root.createNestedObject(F("ForceOffUsermod"));
  obj[F("enabled")] = enabled;
  obj[F("intervalSeconds")] = intervalSeconds;
}

bool UsermodForceOffIfOff::readFromConfig(JsonObject& root) {
  JsonObject obj = root[F("ForceOffUsermod")];
  if (obj.isNull()) return false;

  bool oldEnabled = enabled;
  int oldInterval = intervalSeconds;

  enabled = obj[F("enabled")] | true;
  intervalSeconds = obj[F("intervalSeconds")] | 60;
  intervalSeconds = max(5, intervalSeconds); // 最小5秒，避免误操作

  intervalMs = intervalSeconds * 1000;

  return (enabled != oldEnabled) || (intervalSeconds != oldInterval);
}

void UsermodForceOffIfOff::addToJsonInfo(JsonObject& root) {
  JsonObject info = root.createNestedObject(F("ForceOffUsermod"));
  info[F("enabled")] = enabled;
  info[F("intervalSeconds")] = intervalSeconds;
}
```

then I turn off LEDs, manually trigger a noise signal to light up the first several LEDs.

I could see the log `[ForceOffUsermod] Forced off (bri == 0)`, but the LEDs DO NOT turn off.

How could I turn them off? (What's the right function to call?)

DedeHai, 2025-04-18:
and the built in off refresh does not work?

Fmajor, 2025-04-18:
> and the built in off refresh does not work?

thanks, I know this config for the first time.
open it and now I can not even trigger a interference signal.
problem solved.

closed 2025-04-18 by Fmajor