wled/WLED #4655: ESP-NOW not functioning on latest build

A6uh, 2025-04-22:
### What happened?

Wizmote not being detected on latest build. Manually inputting the known Wizmote address also doesn’t yield any results when pressing buttons.

### To Reproduce Bug

Flash WLED 0.16.0-alpha (build 2412040) to ESP-32
Attempt to pair ESP-NOW remote

### Expected Behavior

It should be detecting the ESP-NOW remote address, and after pairing, should respond to button input.

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.16.0-alpha (build 2412040)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Relevant log/trace output

```shell

```

### Anything else?

Reverting to WLED 0.15 release resolves the issue.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

DedeHai, 2025-04-23:
works perfectly fine. make sure you have wifi sleep disabled.

A6uh, 2025-04-23:
Could you double check this with a freshly flashed ESP32? It appears that ESP-NOW only doesn't work if you don't connect the ESP to WiFi. Like connect to the WLED AP, and try to setup the remote. If you connect the ESP to WiFi, ESP-NOW functions correctly.

Unfortunately for the project I'm trying to use it for, it won't be anywhere near a WiFi network to connect to.

A6uh, 2025-04-23:
Here are some logs I collected. ESP-NOW was enabled and wifi sleep was disabled by default. When I connect to a wifi network is on line 446.

https://pastebin.com/jUeNbRmz

DedeHai, 2025-04-24:
interesting find, not sure that is a bug or intentional or a HW limitation. Did this ever work in older versions?

blazoncek, 2025-04-24:
@A6uh how far have you researched ESP-NOW? It will help you understand where the problem is after you do that.

Brief recap: As WiFi and ESP-NOW share same circuitry they can't work independently. If you want uninterrupted ESP-NOW packets between two ESP devices they have to:

1. be connected to the same WiFi channel
2. must have WiFi disabled and still use same channel for ESP-NOW
3. be in AP mode on the same channel

Once these are fulfilled, ESP-NOW will work as expected. WLED is designed to operate either in 1) or in 3) mode.

I've been working on correct ESP-NOW implementation for over a year now and best take is available in my fork but it may still have quirks.

As for Wizmote: It is an interesting beast. It uses ESP32 inside which will broadcast (proprietary) message on every possible channel in brief bursts. This allows any ESP device to pick ESP-NOW packet on the channel it is currently connected (or configured) to. However if a device has WiFi configured but is not connected to it will switch channels several times per second while searching for WiFi SSID it is configured to. This makes working ESP-NOW difficult as channels change very often.

A6uh, 2025-04-24:
#3 is actually where I think the issue is. If you look at line 451 in those logs, ESP-NOW doesn't actually enter AP mode at all until you try connecting to a WiFi network. Up until then, it doesn't initialize. Shortly after on 487 it stops ESP-NOW, then reenables it in STA mode on 494. I assume STA mode is #1.

I figure it should have been initializing ESP-NOW's AP mode once it was enabled in the UI.

Oh and to answer @DedeHai 's question, yes it did work correctly on 0.15. That's how I originally solved it in the meantime.

blazoncek, 2025-04-24:
Dare to take my fork for a spin? Use stable branch.

A6uh, 2025-04-24:
It looks like your fork kind of has the same issue. Once ESP-NOW is enabled in the UI, ESP-NOW isn't initialized and put into AP mode. However, if you restart the ESP32 afterwards, it puts ESP-NOW into AP mode while it's starting up. Then it works great!

Unfortunately just restarting the ESP32 doesn't fix it for WLED 0.16.0

blazoncek, 2025-04-24:
> Once ESP-NOW is enabled in the UI, ESP-NOW isn't initialized and put into AP mode

That is "normal" and as "designed". I'm not saying that it is correct.
However, ESP-NOW in my fork is designed to work in "temporary AP mode".

DedeHai, 2025-08-31:
@Arcitec is this issue resolved or is there any PR that will resolve it?

Arcitec, 2025-08-31:
@DedeHai ESP-NOW is stable together with WiFi on WLED 0.15.2. The issue with disconnection of the remote only happens if the user's controller has an Ethernet port which is enabled at the same time as a WiFi network SSID is configured.

Apparently the 0.16/main code has changed things and is more broken? I haven't been checking the 0.16 code, since Blaz is working on a rewrite of that code. I've been waiting on that to be merged to main so I can port my ESP-NOW + Ethernet fixes to the new system.

DedeHai, 2025-08-31:
@Arcitec I thought you mentioned the issue somewhere in the lengthy discussions: in AP only mode, ESP-Now code is skipped, the solution proposed above is to enter a bogus SSID so the ESP-Now code gets executed. So in short, the actual issue is known, I was just wondering if there is already a solution in progress.

Arcitec, 2025-08-31:
@DedeHai Oh yeah thanks for the reminder. It's been a few months since we had that part of the discussions. If the user hasn't configured the controller to connect to either a WiFi network or connected an Ethernet cable, then I think it won't initialize the ESP-NOW code, since that's done as part of the "connect to a network" handling. Putting any random SSID into the network config will make WLED attempt to find that network and then initialize ESP-NOW.

And yeah as you say, Blaz has been working on some code to initialize ESP-NOW even when the controller doesn't connect to any WiFi network. I remember hearing about that.

The state of the networking code in WLED is very bad. It's an intermixed spaghetti mess across half a dozen different files that all compete for control over the network. Various configuration handlers abruptly kill the network, force reinitializations, etc. There's no separation of concerns. Everything just writes directly to the state and messes things up and competes for control. It's crazy to say the least. So I'm looking forward to Blaz's improvements, no matter what they might be. I've got a decent solution for the bigger Ethernet + WiFi + ESP-NOW conflict in 0.15.x's codebase (where they don't work together at all), but since a lot of the network code is being changed, I've been waiting to see what happens first.

DedeHai, 2025-09-01:
@blazoncek is there an easy code change that would allow ESPNow to be initialized even if there is not wifi credentials configured? If yes, can we add that to 0.16 and if needed to 0.15.2?

blazoncek, 2025-09-02:
Don't hold your breath for my ESP-NOW code. It has been put on hold as priorities have changed.

DedeHai, 2025-09-02:
No worries.
I found where this bug was added:  https://github.com/wled/WLED/commit/aed03cd03baca7100f7eef713905bc3bae6b55ce
part of https://github.com/wled/WLED/pull/4495 
@netmindz this line was added to fix a DMX input crash, see commit above:
https://github.com/wled/WLED/blob/87092ccb80683ab2022ac0930f8ff8ced1de6033/wled00/wled.cpp#L652
is there a better way to fix the issue mentioned in the commit instead of returning early and with it disable ESPNow in AP only mode?