wled/WLED #4656: Oscillate effect has no fxdata

resnbl, 2025-04-22:
### What happened?

The data fetched from `json/fxdata` for the Oscillate effect is the empty string, despite the effect making use of several parameters. I believe this is due to `FX.cpp`, line 1850 (as of this writing) being:
```  static const char _data_FX_MODE_OSCILLATE[] PROGMEM = "Oscillate";  ```
The string should end with an '@' plus fxdata info. Compare with the Lightning effect just below this:
```  static const char _data_FX_MODE_LIGHTNING[] PROGMEM = "Lightning@!,!,,,,,Overlay;!,!;!";  ```

### To Reproduce Bug

Query WLED to download via: `.../json/fxdata`
The 62nd entry will contain "" instead of valid fxdata

### Expected Behavior

The resulting string should be something like:
"!,!;1,2,3;!"
(But don't quote me - I may not have encoded this correctly or completely.)

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.15.0 Build 2412100

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Relevant log/trace output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

resnbl, 2025-04-23:
Looks like the Solid effect could use some love, too. It also is defined as "", but it uses the Fx color setting. According to the docs: 
```The fallback value if this section is missing is 3 colors: Fx + Bg + Cs.```
Likewise for the Palette section:
```The fallback value if this section is missing is palette selection enabled.```
but Solid doesn't use a palette.

DedeHai, 2025-04-23:
while there could be parameters added, the effects and their UI work perfectly fine as-is.
if you have any suggestions for improvements, please test them and submit a pull request.

blazoncek, 2025-04-23:
Missing metadata means just that effect does not need any external configuration or defaults. There is no requirement for metadata to contain anything but the effect name.

resnbl, 2025-04-23:
@DedeHai Interesting you should mention the UI. I finally perused some of that code (.js is not my forte) and found that in `index.js`, the fxdata loading routine includes:
```js
	// add default value for Solid
	fxdata.shift()
	fxdata.unshift(";!;");
```
Would this have been necessary if the Solid fxdata definition had been included in `FX.cpp`?

My goal is to write a "documentation" program for myself that prints out the configurations of my WLED instances including their presets, playlists, etc. I am attempting to use the fxdata to display the names of settings used and their values. I had been simply working from the documentation, but maybe I should "bite the bullet" and read enough of the UI code to see how it handles the fxdata provided and try to emulate that in my code...