wled/WLED #4662: with WLED_DEBUG and USB_CDC the UART TX pin gets blocked, although unused/not required for debugging

mazzoo, 2025-04-24:
### What happened?

when I enable debug through `-DWLED_DEBUG` I can't use the pin for `hardwareTX` anymore, although printf-debugs go through USB-CDC (with `-DARDUINO_USB_CDC_ON_BOOT=1`)

in my case where I used I2S for audioreactive, the pin 21 disappeared from being a pin choice in the menu, also the I2S function stopped working, that was on a ESP32-C3 (Super Mini) with
```
  -DUM_AUDIOREACTIVE_ENABLE=1
  -DSR_DMTYPE=1 ; 1=I2S Mic; 4=I2S Mic with MCLK
  -DI2S_SDPIN=21
  -DI2S_WSPIN=10
  -DI2S_CKPIN=20
```

### To Reproduce Bug

the lines related as of b0b3196e52fb617aa5495518c7459da363dabecb seem to be
https://github.com/wled/WLED/blob/main/wled00/xml.cpp#L110
https://github.com/wled/WLED/blob/main/wled00/wled.cpp#L391
https://github.com/wled/WLED/blob/main/wled00/wled.cpp#L454
https://github.com/wled/WLED/blob/main/wled00/wled.cpp#L455

### Expected Behavior

when USB-CDC is used for printf debugging don't allocate UART TX or RX pins. might be in other uses.

### Install Method

Self-Compiled

### What version of WLED?

b0b3196e52fb617aa5495518c7459da363dabecb

### Which microcontroller/board are you seeing the problem on?

ESP32-C3

### Relevant log/trace output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

blazoncek, 2025-04-25:
`hardwareTX` has a value specified by framework (TX macro or global static variable) and is supposed to be used by the `Serial` object. `Serial` is used by default for debug output and since S2, S3 and C3 support both UART and CDC at the same time we leave it to framework to choose the actual device and set TX (see pins_arduino.h).

What that means is that TX (whatever its value is) will be reserved for debug output and will not be available for other use. If you want to use debug with CDC and use UART pins for other use you'll need to modify sources and accommodate that or open an issue with framework to correctly update TX (and RX globals).

Please also experiment with ARDUINO_USB_MODE and ARDUINO_USB_CDC_ON_BOOT.

See also: https://community.platformio.org/t/esp32-c3-framework-arduino-serial-print-usb/30464/2

DedeHai, 2025-04-25:
since we are currently stuck on that framework, is it possible to implement a workaround?

mazzoo, 2025-04-25:
why not as simple as this?
```
diff --git a/wled00/wled.cpp b/wled00/wled.cpp
index cc338d23..46b79df9 100644
--- a/wled00/wled.cpp
+++ b/wled00/wled.cpp
@@ -388,8 +388,10 @@ void WLED::setup()
 #endif
 
 #if defined(WLED_DEBUG) && !defined(WLED_DEBUG_HOST)
+#ifndef ARDUINO_USB_CDC_ON_BOOT
   PinManager::allocatePin(hardwareTX, true, PinOwner::DebugOut); // TX (GPIO1 on ESP32) reserved for debug output
 #endif
+#endif
 #ifdef WLED_ENABLE_DMX //reserve GPIO2 as hardcoded DMX pin
   PinManager::allocatePin(2, true, PinOwner::DMX);
 #endif
@@ -451,8 +453,10 @@ void WLED::setup()
   findWiFi(true);      // start scanning for available WiFi-s
 
   // all GPIOs are allocated at this point
+#ifndef ARDUINO_USB_CDC_ON_BOOT
   serialCanRX = !PinManager::isPinAllocated(hardwareRX); // Serial RX pin (GPIO 3 on ESP32 and ESP8266)
   serialCanTX = !PinManager::isPinAllocated(hardwareTX) || PinManager::getPinOwner(hardwareTX) == PinOwner::DebugOut; // Serial TX pin (GPIO 1 on ESP32 and ESP8266)
+#endif
 
   #ifdef WLED_ENABLE_ADALIGHT
   //Serial RX (Adalight, Improv, Serial JSON) only possible if GPIO3 unused
diff --git a/wled00/xml.cpp b/wled00/xml.cpp
index de2f5590..808ad30a 100644
--- a/wled00/xml.cpp
+++ b/wled00/xml.cpp
@@ -107,7 +107,9 @@ void appendGPIOinfo(Print& settingsScript) {
   #endif
   #if defined(WLED_DEBUG) && !defined(WLED_DEBUG_HOST)
   if (!firstPin) settingsScript.print(',');
+#ifndef ARDUINO_USB_CDC_ON_BOOT
   settingsScript.print(hardwareTX); // debug output (TX) pin
+#endif
   firstPin = false;
   #endif
   #if defined(ARDUINO_ARCH_ESP32) && defined(WLED_USE_ETHERNET)
```

if some scenarios prefer HW pins and USBCDC then we could introduce another define DEBUG_TO_UART_TX or similar?

closed 2025-08-31 by github-actions[bot]