wled/WLED #4671: Resolve mDNS address for MQTT broker

schildbach, 2025-05-01:
When connecting to an MQTT broker, please allow putting an mDNS domain into the "broker" field. Currently (on version 0.15.1.beta2) it seems only IP addresses and regular DNS names are supported.

blazoncek, 2025-05-01:
MQTT broker is considered a "server" and as such should have a static IP address.
Adding ability to enter FQDN (or at least hotname) adds to complexity and increases code size.

schildbach, 2025-05-01:
Just for reference, both mDNS and DNS can point to static IP addresses as well.

DNS is already supported in the field.

Tasmota can resolve both mDNS and DNS as well.

Jason2866, 2025-05-01:
Agree that servers have in general a static IP. Nevertheless even in my LAN i don't put an IP address in any config. My mqtt broker has mqtt.lan as DNS name for example. Don't agree that it puts a lot of code to resolve the DNS name in an IP address. It makes it a lot easier to maintain stuff. If the IP address changes of a server it is a lot of "fun" to change all the addresses in all devices. Imho it is a no go **not** to support DNS name resolution.
A lot of flash space can be saved to drop support for the crappy mDNS which brings as "benefit" a lot of issues when used with an esp8266 / esp32.

blazoncek, 2025-05-01:
I did not say it won't or shouldn't be implemented. 😄

schildbach, 2025-05-01:
> Tasmota does not support mDNS

It does: https://github.com/arendst/Tasmota/pull/6866#issuecomment-569732402

blazoncek, 2025-05-02:
As an Apple user I have yet to find an issue with mDNS. My experience is that incorrectly set up networks may exhibit issues with mDNS. And once those are resolved it is up to devices to support it.

That being said, I fully understand the benefits of DNS (or mDNS) resolution. Unfortunately the issue #4586 will add headeaches if only DNS is used, and I have no idea on how to resolve that.

As for WLED to support DNS/mDNS resolution, plenty of UI fields would need to be updated adding code (JS/HTML) as well as backend validation and DNS resolution. If someone finds the time and will to comb through all of it I will be more than happy to help.

schildbach, 2025-07-14:
> [@schildbach](https://github.com/schildbach) please test [#4769](https://github.com/wled/WLED/pull/4769) and report back (in PR).

Is there any way to build WLED in a container, like within Docker or Podman? Or do you have WLED images for pull requests somewhere? I don't have the dev environment available, in particular no Visual Studio.

blazoncek, 2025-07-14:
If you don't need customised binary, please use Github's CI builds. You can find them in PR.
Otherwise follow instructions in [KB](https://kno.wled.ge/advanced/compiling-wled/). AFAIK VSC is free but is not needed. You do need PlatformIO, though.

closed 2025-08-02 by blazoncek