wled/WLED #4674: ESP32 Crashes when controlled over Adalight/TPM2

LucaCraft89, 2025-05-02:
### What happened?

using openrgb to control my WLED esp32 wroom over serial using a Ft232RL serial adapter (because the integrated CP2101 of the esp32 dos not work with openrgb) i can control it fine but after literally 5 seconds of it working and receiving  data it just stops and completely crashes the esp32

### To Reproduce Bug

Use any version of OpenRGB configured as follows to control a ESP32 Wroom over Serial on Windows

[image]

### Expected Behavior

ESP32 does not crash when being controlled over serial.

### Install Method

Binary from WLED.me

### What version of WLED?

0.14.4, because 0.15.0 just does not work over serial

### Which microcontroller/board are you seeing the problem on?

ESP32

### Relevant log/trace output

```shell
None, no errors
```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Szabolcs2008, 2025-05-26:
Same here with an esp32-c3 on the built in usb serial, my wled version is also 0.14.4 (also tried with 0.14.0, still crashed)

DenSwitch, 2025-08-02:
same here esp32 wroom, on any version of wled using built-in usb serial

blazoncek, 2025-08-02:
If anyone wants to be helpful, please provide a crash dump using exception decoder.

DenSwitch, 2025-08-03:
> If anyone wants to be helpful, please provide a crash dump using exception decoder.

are there any instructions? i don't know how i can do that