wled/WLED #4675: Installed 0.16.0-alpha not newest version in Homeassistant

sasukebinbin, 2025-05-03:
### What happened?

Installed
[WLED_0.16.0-alpha_ESP8266.bin](https://github.com/wled/WLED/releases/download/nightly/WLED_0.16.0-alpha_ESP8266.bin) 
from manual OTA update.
But HA report update to nightly as latest version. And it failed to update via UI.

### To Reproduce Bug

Manual OTA update to 0.16.0-alpha

### Expected Behavior

It should be the latest version

### Install Method

Binary from WLED.me

### What version of WLED?

Nightly Release 20250503

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Relevant log/trace output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

netmindz, 2025-05-10:
This is a bug with home assistant and it's WLED integration not WLED.

I've got a few open PRs with the author to help try and fix but they aren't being very responsive

sasukebinbin, 2025-05-12:
Appreciate for the reply. My mistake.

closed 2025-05-12 by sasukebinbin