wled/WLED #4679: WLED Read Only Mode

jmg1970, 2025-05-08:
I've created a wled lamp, I'm also intending to run wled lights outside in the front/back yards.
My issue is I don't want to give the wled ip addresses out to family, because there's no way of keeping the settings safe, anyone who connects to it can change any setting...  unless I'm missing something.  I installed HA and set up a dashboard to turn the lights on and off and activate presets etc, but right now the HA  integration doesn't work properly with wled as it becomes unresponsive very quickly.  Which comes back to an easier solution by some how creating a read only version some how.  possibly using the default web server instance as a read only server showing all the presets and color wheel etc, and then a /admin page for modifications, or simply a user/admin account structure... It maybe impossible, but it would be nice.

blazoncek, 2025-05-08:
Use PIN or OTA lock to prevent any change to settings or firmware.
Presets are another thing but they can be backed up and restored easily.

closed 2025-05-08 by blazoncek

jmg1970, 2025-05-09:
locking the firmware is not what i was asking for!, im fully aware of that.  there are MANY instances where locking the presets etc, is a GOOD idea, share houses, families or just households where there is more than 1 person living there.  Why would locking presets not be a good idea, just telling everyone to just not touch settings when they are turning on a lamp is not good enough, kids and curious/bored adults will always go searching to see what else it can do.  Just backup and restore is stupid with multiple people in a house.  Bare minimum you could help with HA or something to get the integration more stable and usable.

blazoncek, 2025-05-09:
That will not happen I'm afraid (at least it is very unlikely).
There is a script available that will back up presets and it is possible to restore presets with a scheduler from a PC.

Let your children and elderly explore and be creative. If you find that annoying, schedule nightly restore of presets.