wled/WLED #4684: UDP Sync

Electromag1888, 2025-05-12:
### What happened?

All WLED units are using ESP32.  I have one unit for outside Landscape lighting and two units for cabinet lighting.  One cabinet unit is synced to the other cabinet lights via group 4 with Art-Net RGBW.  We will call this unit the Main Cabinet Unit.  This same unit is also set under group one to receive UDP from the Landscape lights.  Ok, so the Landscape lighting is also set to transmit in group one.  The operation should be that whatever the landscape lights are showing should be sent to the main cabinet unit.   The main cabinet controller should send the UDP data to the second cabinet controller as well, therefore all three controllers should be displaying the same.  The problem is that when all lights are on and I change the landscape lights, the main cabinet controller changes colors as it should, however the second cabinet does not.  The second cabinet only changes if I turn off the main cabinet lights and then turn them back on, so the UDP data is only transmitted upon a direct change.  How can I get the main cabinet controller to send updated UDP data to the second cabinet without having to cycle the power off and then back on??  Just to recap, If the landscape lights are on and I change their color and THEN turn on the main cabinet controller, both cabinet controllers turn on display the correct color/effect.  If the landscape and cabinet controller are all on and I change the landscape color (or the scheduler changes the landscape color) then only the main cabinet controller changes and the second cabinet controller stays the previous color/effect.... until I turn off the main controller and then back on.  Then both cabinet controllers display the same color/effect.

### To Reproduce Bug

Instead of using Art-Net RGBW, I was actually using DDP RGBW and it was working perfectly for months, however for some reason this started: when I turn the main cabinet controller to off, the second cabinet controller would turn it's lights off just as it should, however after 2.5 seconds (E1.31 timeout) the second cabinet controller turns the lights back on.  I have fully reset the ESP32 back to factory settings and set everything back up using my backup configuration, however the lights still turn back on after being turned off.   This is why I tried switching to Art-Net RGBW which works other than the second cabinet lights do not get updated until you turn the main cabinet controller off and then on.  Hope this makes sense.  Thank you for any help.

### Expected Behavior

What was working with DDP RGBW is this:  Status- all units activated and lights are on.  When the landscape unit's light color is changed, the main cabinet receives color/effects data through group 1.  The main cabinet unit would then send the data via group 4 color/effects to the second cabinet controller and all three units were color matched.  Landscape sync group: send group 1, Cabinet main sync group: receive group 1 - color, effects, send group 4.  Cabinet second unit sync group: receive group 4 - brightness, color, effects.

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.15.0-B5

### Which microcontroller/board are you seeing the problem on?

ESP32

### Relevant log/trace output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

blazoncek, 2025-05-12:
Did any of this work in 0.13 or 0.14, if not I would not call it a bug.

Electromag1888, 2025-05-13:
> Did any of this work in 0.13 or 0.14, if not I would not call it a bug.

It was working with the version I posted and now it's not.  I would call that a bug regardless of previous versions.

blazoncek, 2025-05-14:
FYI Beta features might get removed.