wled/WLED #4685: Sync function to sync all WLED devices not working

xlachox-official, 2025-05-12:
### What happened?

Since WLED version 0.15.0, using an ESP8266 (bought from AliExpress; manufacturer model is D1 Mini ESP-12F), the sync function in the WLED app has not been working. I cannot sync all my WLED devices at once to change the LEDs' color in my rooms, for example. Another issue is that I often cannot access my WLED devices in the app; no connection to the devices is established. Rebooting does not fix this issue. Sometimes, Alexa does not recognize my WLED devices and says they are offline. I hope you can fix this bug. The WLED project is amazing; thank you for your time!

### To Reproduce Bug

Since WLED version 0.15.0, using an ESP8266 (bought from AliExpress; manufacturer model is D1 Mini ESP-12F), the sync function in the WLED app has not been working. I cannot sync all my WLED devices at once to change the LEDs' color in my rooms, for example. Another issue is that I often cannot access my WLED devices in the app; no connection to the devices is established. Rebooting does not fix this issue. Sometimes, Alexa does not recognize my WLED devices and says they are offline. I hope you can fix this bug. The WLED project is amazing; thank you for your time!

### Expected Behavior

Since WLED version 0.15.0, using an ESP8266 (bought from AliExpress; manufacturer model is D1 Mini ESP-12F), the sync function in the WLED app has not been working. I cannot sync all my WLED devices at once to change the LEDs' color in my rooms, for example. Another issue is that I often cannot access my WLED devices in the app; no connection to the devices is established. Rebooting does not fix this issue. Sometimes, Alexa does not recognize my WLED devices and says they are offline. I hope you can fix this bug. The WLED project is amazing; thank you for your time!

### Install Method

Self-Compiled

### What version of WLED?

WLED 0.15.0

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Relevant log/trace output

```shell

```

### Anything else?

Since WLED version 0.15.0, using an ESP8266 (bought from AliExpress; manufacturer model is D1 Mini ESP-12F), the sync function in the WLED app has not been working. I cannot sync all my WLED devices at once to change the LEDs' color in my rooms, for example. Another issue is that I often cannot access my WLED devices in the app; no connection to the devices is established. Rebooting does not fix this issue. Sometimes, Alexa does not recognize my WLED devices and says they are offline. I hope you can fix this bug. The WLED project is amazing; thank you for your time!

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

blazoncek, 2025-05-12:
Please set sync events (i.e. direct change) in Sync settings.

xlachox-official, 2025-05-12:
> Please set sync events (i.e. direct change) in Sync settings.

Thank you for your information; I will try it out :)