wled/WLED #4689: usermods/PWM_fan I want to be able to invert the PWM

MikhailTeddy, 2025-05-13:
Thanks to everyone who worked on this project!

I use the user mode PWM_fan together with a server fan in the power supply.
This fan is controlled by reverse PWM.
I set both the minimum and maximum speed to 90%, and there is no noise. But because of the reverse PWM, the functionality with temperature is not available to me.

I would like to be able to invert the PWM, just like in esphome.

MikhailTeddy, 2025-05-17:
Added in usermod_PWM_fan.h to line 201 (to the end of the setup function). GPIO.func_out_sel_cfg[pwmPin].inv_sel = 1;
This fixed manual control for me.
But automatic control didn't work for me, the fan always runs at "Max PWM Percent" speed.
My sensor is bme280, maybe that's the problem.
I can't do it without professionals, I'll wait, maybe someone will finish this usermod.

DedeHai, 2025-05-17:
thats not the way to do it.
just invert `pwmValue` i.e. `ledcWrite(pwmChannel, 255 - pwmValue);`

MikhailTeddy, 2025-05-17:
Thank you! Better and easier!
I spent a day to figure out how to invert the pin :)
But the temperature is still the same. The fan runs at a speed of "Max PWM Percent"