wled/WLED #4691: 2d configuration will not persist through reboot when ledmap.json is loaded.

marknelson, 2025-05-17:
### What happened?

I have a HW setup that requires ledmap.json to properly configure a 2d matrix. When ledmap.json is loaded, I have a 2d configuration that is simple: 32x32, no offsets, no serpentine, and all 2d effects look correct. I automatically get a single segment with start/stop of 0/32 and 0/32.

After rebooting, my single segment has invalid values of 0/1024 and 0/1, making it look like it is still a 2D configuration but not a usable one.

I've tried this on quite a few releases, including the base ESP32 0.15.0, which is what I have loaded now. I've been testing with an Adafruit Sparkle Mini and a Bincomini generic ESP32 device. All of this happens identically on both types of hardware.

I've include a config and presets that I used with no ledmap file - it more or less works with my weird LED layout but 2d effects don't seem to work at all. I am suspicious of the ledmap file - it seems like the system is pretty sensitive to formatting, *and* to when and how I upload the new copy, but there are no testing opportunities other than works/doesn't work as far as I can tell. (The system also seems sensitive to when I actually upload the file but I can't pin this down.)

I saved cfg before and after reboot - identical. I have tried to save information by having a saved preset, but I don't see any difference.

### To Reproduce Bug

Configure a system with a 2d matrix using an ledmap.json file. Verify that the ledmap.json is actually being used, maybe by inverting the array. Verify 2d effects are working. Verify that segment shows proper geomerty. Reboot system and check to see that segment geometry is now not good.

### Expected Behavior

Expect 2d matrix setup to be intact after reboot.

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.15.0 (build 2412100). (have tested with many other builds)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Relevant log/trace output

```shell

```

### Anything else?

All minified files were prettified for easier viewing - but are not in use on the device.

The first two files are from a setup without ledmap.json - it doesn't exactly work because my LEDs are not in a conforming geometry, but the segments and 2d matrix survive reboot.

[cfg_noledmap.json](https://github.com/user-attachments/files/20266209/cfg_noledmap.json)
[presets_noledmap.json](https://github.com/user-attachments/files/20266210/presets_noledmap.json)

This is my config when using ledmap.json. I was going to include cfgs and presets from before and after reboot, but they are identical.

[cfg_before_reboot.json](https://github.com/user-attachments/files/20266215/cfg_before_reboot.json)
[ledmap.json](https://github.com/user-attachments/files/20266216/ledmap.json)
[presets_before_reboot.json](https://github.com/user-attachments/files/20266218/presets_before_reboot.json)

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

DedeHai, 2025-05-18:
I tested what you described to reproduce, and led map works even after reboot.

blazoncek, 2025-05-18:
Please show the info page dialog. I suspect FS issue.

marknelson, 2025-05-18:
Yeah FS issue is a reasonable guess for configuration not surviving a reboot. The one thing going against it is that I see identical behavior on two completely different pieces of hardware. Maybe a second indication is no other flakiness of any sort with config files or anything else for that matter.

Something funny is going on though. Even though my ledmap.json works, when I upload it I often get this 500 message which unfortunately includes no context:

[image]

Not good, for sure. FYI I have not done it in the past but I am testing this morning after a factory reset with minified ledmap.json, no noticeable difference, after reboot ledmap is good but 2d segment is hosed.

Here is the info page:

[image]

blazoncek, 2025-05-18:
FS is ok by that picture (983k).
FYI Ledmap for 2D needs 2D format.

marknelson, 2025-05-18:
Ah that makes total sense, thanks! Is there a doc page or an example I can follow? I actually have the data in 2d format already, in a spreadsheet of all things, just need formatting help. I am guessing:

{"map":[
  [R0C0,R0C1,R0C2...
  [R1C0,R1C1,R1C2...
  ...
  [RNC0, RNC1,...RNCN]
}
}

?

blazoncek, 2025-05-18:
Wrong.
Needs "width" and "height" keys.

See: https://kno.wled.ge/advanced/mapping/#how-do-we-do-it

closed 2025-05-18 by blazoncek