wled/WLED #4699: Support ESP32-S3 or refuse to flash it

KE55ARD, 2025-05-24:
*Issue*
People keep trying to flash my LED controller using the WLED web flasher and bricking them because it doesn't use the correct build.

*Suggestion*
Please either build the required version for the ESP32-S3 variant, or refuse to flash it so that they can't brick them.

*Context*
I sell an addressable LED controller which comes with it's own software, but most people install WLED on it instead. And despite having instructions on how to do this, they all just go to the WLED web flasher and use that, which then bricks them with the wrong build.

KE55ARD, 2025-05-24:
Thanks for responding.

To clarify when I used the term "bricked" I meant from the perspective of the average user. Most of whom seem to just about know how to flash them from a browser. But once the incorrect software is flashed they don't connect to a browser reliably enough to flash them again.

I do also provide instructions on finding and flashing a working build, but people seem to think they know better by rushing to your flashing tool.

I understand that if there are too many variants to support them all that's fair enough. But could you at least identify an unsupported variants and avoid attempting to flash them?

lost-hope, 2025-05-26:
Only thing the web flasher can check is the chip type (s2,S3,c3,plain,8266).
It can't detect size or flash type.

KE55ARD, 2025-05-26:
Ok, so it *could* detect that it's an S3 and refuse to flash. Because it knows that it can't get the correct build without more information?

blazoncek, 2025-05-27:
We fill not refuse to flash as quite a few boards *can* be flashed successfully by selecting correct binary.
What we could do is list those boards in the KB. But that's the task for WLED-Docs.

revk, 2025-05-27:
I think he means refuse to flash the INCORRECT binary, so yes, if you select the correct binary, great.

I.e. if you have selected a binary for something other than an ESP32-S3 and find it is an ESP32-S3 connects, at the very least warn it is wrong, and ideally refuse to flash the wrong binary.

As for bricked, the board in question does not have GPIO0 or serial tx/rx exposed, only USB. USB usually works fine, but for some reason when loaded with the wrong binary it goes in a reboot loop that means connecting USB at that point is very hit and miss.

I hope that helps.

netmindz, 2025-07-06:
Let me know @lost-hope if I can be any help with the check that chip type matches chip type of selected option

lost-hope, 2025-07-06:
The Installer can not detect anything except chip type. No flash size nor flash type. 
ATM the only S3 binary served is 4mb qspi. 
to make the support for most chips easier, we need to change the binaries in the releases. ( see https://esphome.github.io/esp-web-tools/ "preparing your firmware" )
or adding a shit ton of different binaries and checkboxes to build the binaries. (and that will always be a huge manual work on each update)

DedeHai, 2025-07-06:
Adding several S3 builds to default bin builds should not be an issue right? 
Would adding a dropdown with chip compatible bins after chip detection be a viable solution? I.e. if S3 is detected, ask the user to select flash/ram config?

spiro-c, 2025-07-11:
@lost-hope Preparing the merged version of the esp32 what have proper version of the bootloader partition and flush mode can be done easy on during the build process in pio [Espeasy](https://github.com/letscontrolit/ESPEasy/blob/mega/tools/pio/post_esp32.py)  [Tsmota](https://github.com/arendst/Tasmota/blob/development/pio-tools/post_esp32.py) and other firmware's are doing this also the manifest for the  espweb tools can be generated on the fly https://github.com/esphome/esp-web-tools/issues/498#issuecomment-2989602603

keppycs, 2025-07-27:
Is full S3 support on the roadmap? What's the status on this?

blazoncek, 2025-07-27:
S3 is fully supported in firmware.
Flashing it is tricky, however.

robhybrid, 2025-09-01:
I'm having the same issue. The web installer is trying to download:

https://github.com/wled/WLED/releases/download/v0.15.1/WLED_0.15.1_ESP32-S3_8MB.bin

but the file does not exist. It looks like the file name was changed to:

https://github.com/wled/WLED/releases/download/v0.15.1/WLED_0.15.1_ESP32-S3_8MB_opi.bin

Note: there is an **_opi** extension added, and I'm not sure what that means.
Someone forgot to tell the web installer. Either file moved, or maybe there is no S3 support in this release. There is defacto no web-installer support for the S3 in this release.

Reading the source code I found this in wled00/pin_manager.cpp:
```
    if (gpio > 21 && gpio < 33) return false;     // 22 to 32: not connected + SPI FLASH
    #if CONFIG_ESPTOOLPY_FLASHMODE_OPI            // 33-37: never available if using _octal_ Flash (opi_opi)
    if (gpio > 32 && gpio < 38) return false;
    #endif
    #if CONFIG_SPIRAM_MODE_OCT                    // 33-37: not available if using _octal_ PSRAM (qio_opi), but free to use on _quad_ PSRAM (qio_qspi)
    if (gpio > 32 && gpio < 38) return !psramFound();
    #endif
```
https://github.com/wled/WLED/blob/a037d99469cdd1b52f27e6a584fbe25642c30514/wled00/pin_manager.cpp#L217

We should probably have integration tests to verify the files are in their expected locations. I'm happy to help resolve this issue, if I can get some direction on this. 

Thanks.

blazoncek, 2025-09-02:
The issue is related to web installer and not firmware itself.
If you are experiencing flashing issues with web installer then please open an issue at web installer's [repository](https://github.com/wled/WLED-Webinstaller).

closed 2025-09-02 by blazoncek