wled/WLED #4704: WLED global brightness overwrites ArtNet brightness

Deadeye5589, 2025-05-30:
### What happened?

Using V16, the global brightness in WLED overwrites the channel data send via ArtNet.
Hence data coming in via ArtNet is not used 1 to 1, instead WLED will limit the brightness accoring to the set global brightness parameter.

### To Reproduce Bug

Simple setup. 
3 PWM channels to drive an RGB LED.
Configure ArtNet as single RGB endpoint in WLED.
Limit global brightness in WLED.
Send DMX Data to WLED device. 
LEDs will not reach full brightness, even if DMX transmitter ist sending 255 as channel value. 

### Expected Behavior

I would expect that ArtNet (DMX) is treated as it is. WLED should no post process the data and limit the values that the DMX transmittter is sending. 

### Install Method

Self-Compiled

### What version of WLED?

V16

### Which microcontroller/board are you seeing the problem on?

ESP32

### Relevant log/trace output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

blazoncek, 2025-05-31:
Please use override.

[image]

closed 2025-05-31 by blazoncek

Deadeye5589, 2025-06-01:
Doh, classical issue of PEBKAC. 
Thank you