wled/WLED #4707: The WROVER firmware does not install on the WROVER-B

KM3L0TT, 2025-05-31:
### What happened?

Hello,
I created a WROVER-B-based PCB (4Mb flash / 8Mb PSRAM).

With the installation via "install.wled.me" it apparently installs the EPS32 "classic" firmware. When I want to install the WROVER firmware, it gives me an update error.
The WROVER firmware may be intended for the WORVER-E module and not the B model?

How to take full advantage of the PSRAM integrated into the WROVER-B module?

Thanks

### To Reproduce Bug

Install WROVER bin on WROVER-B module 

### Expected Behavior

I would like the WROVER-B module to be recognized and take full advantage of the PSRAM embedded on it.

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 15.00

### Which microcontroller/board are you seeing the problem on?

Other

### Relevant log/trace output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

blazoncek, 2025-06-01:
Bootloader for ESP32 with PSRAM is different from the one for ESP without it. Unfortunately web installer cannot distinguish which chip you use.

You will need to compile and flash the ESP yourself using MS VSC + PIO or obtain bootloader, partition map, app0 file and WROVER binary for ESP32.

KM3L0TT, 2025-06-01:
Thank you for your reply.
 I understand better indeed. I'll watch it quietly. 

Thank you again

closed 2025-06-01 by KM3L0TT