wled/WLED #4730: Transition between individually addressed LEDs

GtwoK, 2025-06-13:
I'm using WLED as a way to create an ambient effect behind my TV, dynamically adapting to the album artwork currently being played via Spotify. I'm posting the colours as individually addressed LEDs (ie {'seg': [{'i': [[239, 117, 8], [239, 117, 8], [239, 117, 8], ... ]}]} ), which works fine. However, the transition is instantaneous. Including a transition parameter when defining the segments does not have any effect. I thought based on [#826](https://github.com/wled/WLED/issues/826) that the issue would be fixed, but it seems maybe that only applies to predefined segments? Not entirely clear.

At any rate, it'd be nice be able to specify the transition parameter and have that function when providing an array of light colours. Currently, I'm having to dynamically calculate 50 steps between the before and after state and push them 20ms apart to achieve a smooth transition. This is obviously a computationally heavy thing to do, and also has some caveats for me as far as having to store previous values / not working if the light colours are set externally — something that would be far easier if WLED could calculate it automatically based on the current state.

blazoncek, 2025-06-14:
Individual pixel control is not something WLED takes control of. Once you set a pixel color, WLED immediately forgets that.

If you want transitions, configure segments and use "Solid" effect with appropriate color information. This will use transiton time.

TL;DR: It is impossible to use transitions with individual pixel control.

GtwoK, 2025-06-14:
Gotcha, okay. 

There is a limit of 32 segments isn't there? I know I must be missing something but surely there must be way to have control of every led on my strip (89 total) and have them be able to transition, no? Or else what would be the point of individual control of 5m strips and such?