wled/WLED #4734: PWM RGB+CCT WLED_0.16.0-alpha_ESP32 unable to switch between warm and cool

Siempo2000, 2025-06-16:
### What happened?

I have flashed one of my quinled AN-Deca with WLED_0.16.0-alpha and everything seems to work fine except i cannot switch between warm and cool white. No matter what i do, both warm and cool leds output remain the same when sliding the slider from warm to cool.

On version v0.15.1.beta2 using the same environment below it works as expected.

### To Reproduce Bug

Flash esp32 with VS Code using the following env:

[env:quinled_deca]
extends = esp32dev
board = esp32dev
platform = ${esp32.platform}
upload_speed = 921600
platform_packages = ${esp32.platform_packages}
build_unflags = ${common.build_unflags}
lib_deps = ${esp32.lib_deps}  ${digboards.lib_deps_temperature}
board_build.partitions = ${esp32.default_partitions}
build_flags = ${common.build_flags} ${esp32.build_flags}
  -D WLED_DISABLE_BLYNK
  -D WLED_BRAND="\"QuinLED\""
  -D WLED_PRODUCT_NAME="\"An-Deca\""
  -D SERVERNAME="\"An-Deca\""
  -D WLED_RELEASE_NAME="\"Quin-An-Deca\""
  -D NOWIFISLEEP=false
  -D DATA_PINS=16,17,18,19,21,25,26,27,14,12
  -D LED_TYPES=TYPE_ANALOG_5CH,TYPE_ANALOG_5CH
  -D PIXEL_COUNTS=1,1
  -D ABL_MILLIAMPS_DEFAULT=0
  -D BTNPIN=22
  -D BTNTYPE=BTN_TYPE_PUSH
  -D IRPIN=-1 
  -D I2S_SDPIN=-1
  -D I2S_WSPIN=-1
  -D I2S_CKPIN=-1 
  -D USERMOD_DALLASTEMPERATURE
  -D TEMPERATURE_PIN=32
  -D WLED_AUTOSEGMENTS
custom_usermods = audioreactive Temperature  
monitor_filters = esp32_exception_decoder
extra_scripts = ${scripts_defaults.extra_scripts}

Connect Analog RGB+CCT strip to AN-Deca and turn off color and turn up white level
Next slide the warm/cool slider to warm or cool and observe the warm/cool leds

### Expected Behavior

When i slide the warm/cool slider to warm only the warm led light up, when i slide it it cool only the cool led light up.

### Install Method

Self-Compiled

### What version of WLED?

WLED_0.16.0-alpha 

### Which microcontroller/board are you seeing the problem on?

ESP32

### Relevant log/trace output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

DedeHai, 2025-06-16:
Confirmed. Can't manually set CCT even if not using any auto-calculation for white: 
- Auto-calculate W channel from RGB set to none
- Auto-calculate white set to none
- CCT from RGB disabled

I suspect a bug in the config page or handling thereof: each time I hit save, the value of `CCT additive blending` increases by itself so there is definitely something off with it. I noticed strange things on that config page before, like GPIO numbers getting artefact chars so the bug may also be in java.

blazoncek, 2025-06-18:
The problem is (as I guessed) the fact that CCT is a segment property and since segments can be blended this information is lost during pixel drawing (see lines 1229-1233 in [FX_fcn.cpp](https://github.com/wled/WLED/blob/d2d5c423b768001e38f5cb3637bb8d34cf86b046/wled00/FX_fcn.cpp#L1229-L1233) and blending in [show()](https://github.com/wled/WLED/blob/d2d5c423b768001e38f5cb3637bb8d34cf86b046/wled00/FX_fcn.cpp#L1611)).

Unfortunately I have no idea how to resolve the problem without entire rewrite of pixel processing:
a) include CCT information (or WW & CW) in pixel buffer (increases all buffers by 20% and will incur speed penalty)
b) ignore segment property and use global CCT (reduces functionality)

It is difficult (and likely impossible) to *reverse-lookup* CCT information from pixel position as segments may be overlaid and each segment may have different CCT value. The reverse-lookup will also slow down processing as each pixel needs to be reverse-mapped to one and only one segment.

I am open to ideas. /cc @netmindz @willmmiles @DedeHai

DedeHai, 2025-06-18:
I am still not fully aware on how white and CCT are processed as it still seems convoluted to me. 
Since each segment has a CCT value (and only one, not per pixel) I don't think adding a full buffer is a good way to solve it, but it may be the easiest. Also doing it per bus is not a good solution: this would work for analog RGB-CCT strips but not for addressable one where a user can have differen zones: here a full buffer is needed or reverse-lookup. 
Thoughts:
- could the white channel somehow be "abused" to also store some CCT info in some clever encoding? resolution would not be great.
- adding a seperate 8-bit buffer for CCT and keeping curent 32bit buffer for RGBW could be a solution that only requires more RAM & CPU time if CCT is actually used 
- reverse-lookup feels like not a clean solution

blazoncek, 2025-06-20:
@Siempo2000 please test proposed PR.

Siempo2000, 2025-06-21:
Confirmed to be working now.

PR #4744 (2025-06-24): CCT fix
fixes #4734

## Summary by CodeRabbit

- **New Features**
	- Improved color temperature (CCT) handling for lighting effects, now supporting per-pixel CCT control for more precise and dynamic color rendering.

- **Bug Fixes**
	- Enhanced memory management to ensure proper cleanup of resources used for per-pixel CCT data.