wled/WLED #4739: Ethernet seems to disable ESP-NOW in a cycle

Arcitec, 2025-06-19:
### What happened?

I think there's an issue where Ethernet is causing the WiFi antenna to be turned off, which is potentially what's breaking ESP-NOW.

`WiFi.disconnect(true)` turns off the entire antenna in `network.cpp` whenever Ethernet has a connection event, for example.

### To Reproduce Bug

It can be reproduced on 0.15.x and 0.16.x by just using an Ethernet board and simultaneously enabling ESP-NOW. The ESP-NOW remote will be unreliable (laggy or no response at all).

The workaround is to set "Ethernet Type = None" and restarting the controller so that WLED is in pure WiFi mode and doesn't get confused and doesn't turn off ESP-NOW.

This strongly indicates that the code handling that's Ethernet is turning off the WiFi antenna (and ESP-NOW requires the antenna).

### Expected Behavior

The desired WLED networking state would be to detect when a Ethernet cable is connected, and to then do the following:

- "Ethernet is Connected, so let's..."
  - Disable WLED AP and regular WiFi clients, since Ethernet will take care of the actual *networking*.
  - But still keep the actual WiFi radio enabled for ESP-NOW, if the ESP-NOW feature is enabled.

### Install Method

Self-Compiled

### What version of WLED?

esp32_eth build of 0.15.1 branch, and 0.16.x (main)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Relevant log/trace output

```shell

```

### Anything else?

I began to investigate the issue, but WLED's networking code is a real behemoth of bolted-on complexity, since it probably started with just WiFi support, but nowadays it juggles Ethernet, WiFi antenna (ESP-NOW), WiFi clients, WiFi AP, becoming a WiFi client itself onto a home network, etc.

I see that WLED turns on/off its various Ethernet/WiFi/AP/ESP-NOW modes based on trying to make sense of it all. And when clients connect, it re-evaluates everything each time there's a connection event, and then it yet again turns on/off stuff based on what it thinks is happening.

All of that is implemented as a huge code jumble that's spread between `wled.cpp` and `network.cpp`. With such a fragile structure, it's very hard to grasp the entire connection state handling at a glance and there's a risk of edge-cases where combinations of different features would lead to incorrect states.

It seems like this ESP-NOW issue is such an edge case, where Ethernet is most likely triggering a disabling of the WiFi antenna, or perhaps even a "fight" where it turns on/off repeatedly based on the ESP-NOW code trying to enable the antenna and the Ethernet code trying to disable the antenna within each main `loop()` cycle.

There's a related ticket at #4655 which mentions some other fixes/code improvements for ESP-NOW robustness. But I think that these issues are separate.

I would like to help out, but would need to learn a lot more about this very complex network state handling. Does anyone have any ideas or theories?

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

blazoncek, 2025-06-19:
That was intentional at the time.
However, your recent findings corroborate the decision.
Ethernet & WiFi don't go well together.

Arcitec, 2025-07-31:
@blazoncek I am very sorry for not seeing this earlier, the email missed me. :(

> Ethernet & WiFi don't go well together.

They work perfectly together, but *only* in one situation:

`if (es.eth_clk_mode == ETH_CLOCK_GPIO0_IN) {}`

Then ESP-NOW / WiFi + Ethernet can be used together completely reliably.

That's what I found in the investigation in #4703. Ethernet is corrupted when WiFi is enabled *if* and *only if* the ESP32 is generating and outputting the Ethernet clock signal via its PLL. If the clock signal is external via a crystal (coming into GPIO 0), there is no conflict and WiFi signals are fully compatible with Ethernet usage.

Which means there's currently 3 boards where it works perfectly, and I know that Quindor is releasing an Ethernet-capable board which will use an external crystal precisely so that ESP-NOW remotes will work on his board.

I've been looking into rewriting the network initialization code from scratch to support this scenario, but I'm looking at your fork now and see that you're in the middle of a large rewrite already. Hmm.