wled/WLED #4752: User mod requesting settings be saved to disk via readFromConfig() return value causes crash

BootsSiR, 2025-06-28:
### What happened?

If a user mod returns false to have WLED save defaults to disk, the settings get saved missing segment/bus information and you get stuck in a boot loop with this crash:

```
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400edbd5  PS      : 0x00060e30  A0      : 0x800ee435  A1      : 0x3ffb1dc0  
A2      : 0x00000000  A3      : 0x3ffb89d4  A4      : 0x00000000  A5      : 0x3ffc22ac  
A6      : 0x00000000  A7      : 0x00000001  A8      : 0x00000001  A9      : 0x00ffa000  
A10     : 0x00000001  A11     : 0x00ffa000  A12     : 0x00000000  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0x401c58a0  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

ELF file SHA256: 0000000000000000

Backtrace: 0x400edbd2:0x3ffb1dc0 0x400ee432:0x3ffb1e90 0x400ee655:0x3ffb1ef0 0x401214e8:0x3ffb1f20 0x401218b2:0x3ffb1f90 0x40143979:0x3ffb1fb0 0x4008c92e:0x3ffb1fd0
  #0  0x400edbd2:0x3ffb1dc0 in WS2812FX::blendSegment(Segment const&) const at wled00/src/dependencies/json/ArduinoJson-v6.h:1061
      (inlined by) WS2812FX::blendSegment(Segment const&) const at wled00/FX_fcn.cpp:1521
  #1  0x400ee432:0x3ffb1e90 in WS2812FX::show() at wled00/src/dependencies/json/ArduinoJson-v6.h:1061
  #2  0x400ee655:0x3ffb1ef0 in WS2812FX::service() at wled00/src/dependencies/json/ArduinoJson-v6.h:1061
  #3  0x401214e8:0x3ffb1f20 in WLED::loop() at wled00/wled.cpp:127
  #4  0x401218b2:0x3ffb1f90 in loop() at wled00/wled_main.cpp:23
  #5  0x40143979:0x3ffb1fb0 in loopTask(void*) at C:/Users/steve/.platformio/packages/framework-arduinoespressif32@src-e9b1fbd6563a55e19ddae15e1fc09589/cores/esp32/main.cpp:23
  #6  0x4008c92e:0x3ffb1fd0 in vPortTaskWrapper at /home/cschwinne/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

Rebooting...
```

### To Reproduce Bug

From latest main code

- Enable the user mod EXAMPLE in platformio.ini:

`custom_usermods = audioreactive EXAMPLE`

- Modify usermods\EXAMPLE\usermod_v2_example.cpp:readFromConfig(JsonObject& root) to return false to indicate settings should be saved to disk:

```
    /*
     * readFromConfig() can be used to read back the custom settings you added with addToConfig().
     * This is called by WLED when settings are loaded (currently this only happens immediately after boot, or after saving on the Usermod Settings page)
     * 
     * readFromConfig() is called BEFORE setup(). This means you can use your persistent values in setup() (e.g. pin assignments, buffer sizes),
     * but also that if you want to write persistent values to a dynamic buffer, you'd need to allocate it here instead of in setup.
     * If you don't know what that is, don't fret. It most likely doesn't affect your use case :)
     * 
     * Return true in case the config values returned from Usermod Settings were complete, or false if you'd like WLED to save your defaults to disk (so any missing values are editable in Usermod Settings)
     * 
     * getJsonValue() returns false if the value is missing, or copies the value into the variable provided and returns true if the value is present
     * The configComplete variable is true only if the "exampleUsermod" object and all values are present.  If any values are missing, WLED will know to call addToConfig() to save them
     * 
     * This function is guaranteed to be called on boot, but could also be called every time settings are updated
     */
    
    {
      // default settings values could be set here (or below using the 3-argument getJsonValue()) instead of in the class definition or constructor
      // setting them inside readFromConfig() is slightly more robust, handling the rare but plausible use case of single value being missing after boot (e.g. if the cfg.json was manually edited and a value was removed)

      JsonObject top = root[FPSTR(_name)];

      bool configComplete = !top.isNull();

      configComplete &= getJsonValue(top["great"], userVar0);
      configComplete &= getJsonValue(top["testBool"], testBool);
      configComplete &= getJsonValue(top["testULong"], testULong);
      configComplete &= getJsonValue(top["testFloat"], testFloat);
      configComplete &= getJsonValue(top["testString"], testString);

[... 269 lines cut ...]
PC      : 0x400edbd5  PS      : 0x00060e30  A0      : 0x800ee435  A1      : 0x3ffb1dc0  
A2      : 0x00000000  A3      : 0x3ffb89d4  A4      : 0x00000000  A5      : 0x3ffc22ac  
A6      : 0x00000000  A7      : 0x00000001  A8      : 0x00000001  A9      : 0x00ffa000  
A10     : 0x00000001  A11     : 0x00ffa000  A12     : 0x00000000  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0x401c58a0  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

ELF file SHA256: 0000000000000000

Backtrace: 0x400edbd2:0x3ffb1dc0 0x400ee432:0x3ffb1e90 0x400ee655:0x3ffb1ef0 0x401214e8:0x3ffb1f20 0x401218b2:0x3ffb1f90 0x40143979:0x3ffb1fb0 0x4008c92e:0x3ffb1fd0
  #0  0x400edbd2:0x3ffb1dc0 in WS2812FX::blendSegment(Segment const&) const at wled00/src/dependencies/json/ArduinoJson-v6.h:1061
      (inlined by) WS2812FX::blendSegment(Segment const&) const at wled00/FX_fcn.cpp:1521
  #1  0x400ee432:0x3ffb1e90 in WS2812FX::show() at wled00/src/dependencies/json/ArduinoJson-v6.h:1061
  #2  0x400ee655:0x3ffb1ef0 in WS2812FX::service() at wled00/src/dependencies/json/ArduinoJson-v6.h:1061
  #3  0x401214e8:0x3ffb1f20 in WLED::loop() at wled00/wled.cpp:127
  #4  0x401218b2:0x3ffb1f90 in loop() at wled00/wled_main.cpp:23
  #5  0x40143979:0x3ffb1fb0 in loopTask(void*) at C:/Users/steve/.platformio/packages/framework-arduinoespressif32@src-e9b1fbd6563a55e19ddae15e1fc09589/cores/esp32/main.cpp:23
  #6  0x4008c92e:0x3ffb1fd0 in vPortTaskWrapper at /home/cschwinne/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

Rebooting...
```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

BootsSiR, 2025-06-28:
I think this is the issue:

wled00\cfg.cpp:244 deserializeConfig()
```
      busConfigs.emplace_back(ledType, pins, start, length, colorOrder, reversed, skipFirst, AWmode, freqkHz, maPerLed, maMax);
      doInitBusses = true;  // finalization done in beginStrip()
```

beginStrip() is called at wled00\wled.cpp:441 during startup which is called AFTER the settings would be saved so the buses wouldn't be initialized and wouldn't be saved to settings causing the behaviour I'm seeing.

blazoncek, 2025-06-28:
I always thought automatic saving was a bad idea. 😄

BootsSiR, 2025-06-28:
> I always thought automatic saving was a bad idea. 😄

I patched your fix into my code and it resolves this bug.  Thanks for the quick turnaround.  I was going to fix it when I woke up this morning but you did it while I slept 😅

closed 2025-07-01 by blazoncek